enjoy-digital/litex #1621: Default bios not compiling for tang primer 20k,

L-jr-Bennett, 2023-02-28:
Error when using --with-sdcard or --with-ethernet for default target for Tang Primer 20k board.
`pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/ssp/chk_fail.c:13: undefined reference to `write'
collect2: error: ld returned 1 exit status`

enjoy-digital, 2023-03-02:
I'm not able to reproduce, the issue seems to be related to picolibc compilation. Can you make sure you are using up to date version of the dependencies (`./litex_setup.py --update`) and provide more info on your machine?

bentwire, 2023-03-13:
I'm experiencing this as well.  From a fresh install:
Commands I ran to install:
./litex_setup.py --init --install --user --config full
./litex_setup.py --update
pushd litex-boards/litex_boards/targets/
./sipeed_tang_primer_20k.py --dock standard --with-sdcard --with-video-terminal --with-ethernet --build

It fails here:
`CC       bios.elf
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: ../libc/libc.a(libc_ssp_chk_fail.c.o): in function __chk_fail:
/home/mike/gowin/projects/litex/litex-boards/litex_boards/targets/build/sipeed_tang_primer_20k/software/libc/../../../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/ssp/chk_fail.c:13: undefined reference to write
collect2: error: ld returned 1 exit status
make: *** [/home/mike/gowin/projects/litex/litex/litex/soc/software/bios/Makefile:72: bios.elf] Error 1
rm crt0.o
make: Leaving directory '/home/mike/gowin/projects/litex/litex-boards/litex_boards/targets/build/sipeed_tang_primer_20k/software/bios
Traceback (most recent call last):
  File "/home/mike/gowin/projects/litex/litex-boards/litex_boards/targets/./sipeed_tang_primer_20k.py", line 257, in <module>
    main()
  File "/home/mike/gowin/projects/litex/litex-boards/litex_boards/targets/./sipeed_tang_primer_20k.py", line 246, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/mike/gowin/projects/litex/litex/litex/soc/integration/builder.py", line 354, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/home/mike/gowin/projects/litex/litex/litex/soc/integration/builder.py", line 288, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/mike/gowin/projects/litex/litex-boards/litex_boards/targets/build/sipeed_tang_primer_20k/software/bios', '-f', '/home/mike/gowin/projects/litex/litex/litex/soc/software/bios/Makefile']' returned non-zero exit status 2.
`

enjoy-digital, 2023-07-31:
Fixed with #1738.

closed 2023-07-31 by enjoy-digital