enjoy-digital/litex #1624: Ubuntu 22.04 riscv gcc causes issue

AEW2015, 2023-02-28:
https://github.com/enjoy-digital/litex/blob/2f5481dbb9ff22cff03f21e8e7d418442b778908/litex_setup.py#L353

```
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: ../libc/libc.a(libc_ssp_chk_fail.c.o): in function `__chk_fail':
/home/deck/xup/litex_repos/build/opalkelly_xem8320/software/libc/../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/ssp/chk_fail.c:13: undefined reference to `write'
collect2: error: ld returned 1 exit status
make: *** [/home/deck/xup/litex_repos/litex/litex/soc/software/bios/Makefile:72: bios.elf] Error 1
rm crt0.o
make: Leaving directory '/home/deck/xup/litex_repos/build/opalkelly_xem8320/software/bios'
Traceback (most recent call last):
  File "/home/deck/xup/litex_repos/./litex-boards/litex_boards/targets/opalkelly_xem8320.py", line 158, in <module>
    main()
  File "/home/deck/xup/litex_repos/./litex-boards/litex_boards/targets/opalkelly_xem8320.py", line 151, in main
    builder.build()
  File "/home/deck/xup/litex_repos/litex/litex/soc/integration/builder.py", line 354, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/home/deck/xup/litex_repos/litex/litex/soc/integration/builder.py", line 288, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/deck/xup/litex_repos/build/opalkelly_xem8320/software/bios', '-f', '/home/deck/xup/litex_repos/litex/litex/soc/software/bios/Makefile']' returned non-zero exit status 2.
```

I am using the https://static.dev.sifive.com/dev-tools/freedom-tools/v2020.08/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14.tar.gz instead.

n-kremeris, 2023-03-15:
I have the same issue on digilent_nexys_video+rocket/linuxd; design fails to build when adding --with-sdcard or --with-ethernet
`./litex-boards/litex_boards/targets/digilent_nexys_video.py --build --cpu-type rocket --cpu-variant linuxd --sys-clk-freq 50e6 --with-sdcard`

I get the following error at the bios building step:
```
 CC       bios.elf
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: ../libc/libc.a(libc_ssp_chk_fail.c.o): in function `__chk_fail':
/work/litex/build/digilent_nexys_video/software/libc/../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/ssp/chk_fail.c:13: undefined reference to `write'
collect2: error: ld returned 1 exit status
make: *** [/work/litex/litex/litex/soc/software/bios/Makefile:72: bios.elf] Error 1
rm crt0.o
make: Leaving directory '/work/litex/build/digilent_nexys_video/software/bios'
Traceback (most recent call last):
  File "/work/litex/./litex-boards/litex_boards/targets/digilent_nexys_video.py", line 209, in <module>
    main()
  File "/work/litex/./litex-boards/litex_boards/targets/digilent_nexys_video.py", line 202, in main
    builder.build(**parser.toolchain_argdict)
  File "/work/litex/litex/litex/soc/integration/builder.py", line 354, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/work/litex/litex/litex/soc/integration/builder.py", line 288, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/work/litex/build/digilent_nexys_video/software/bios', '-f', '/work/litex/litex/litex/soc/software/bios/Makefile']' returned non-zero exit status 2.

```

The same happens when using either ```--with-sdcard``` or ```--with-ethernet```. When neither of these two are specified, the build completes successfully.

The system in question is:
Ubuntu 22.04
gcc:  riscv64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
litex version: d6db2be6b63075268502ea36392b1f1bd517fc00 fully updated

gsomlo, 2023-03-16:
On Wed, Mar 15, 2023 at 08:39:33AM -0700, Norbert Kremeris wrote:
> I have the same issue on digilent_nexys_video+rocket/linuxd; design fails to
> build when adding --with-sdcard or --with-ethernet
> ./litex-boards/litex_boards/targets/digilent_nexys_video.py --build --cpu-type
> rocket --cpu-variant linuxd --sys-clk-freq 50e6 --with-sdcard

With apologies in advance, I'm completely swamped for the next week or
two, and can't try to replicate your command line at the moment.

That said, I have some "leading" questions:

1. does ubuntu expect your kernel to have an FPU? If so, are you using
emulation (e.g., BBL) to make that happen in software? I ask this
because the `linux` variant does *not* currently have an fpu in
gateware, so it only implements `rv64imac` (no `fd` in there)!

2. The nexys-video has enough room to build using the `full` variant,
which *does* implement `rv64imafdc` (and a bunch on top of that, more
on that in a second). I have no trouble building something like:

	litex-boards/litex_boards/targets/digilent_nexys_video.py --build \
		--cpu-type rocket --cpu-variant fulld --sys-clk-freq 50e6 \
		--with-ethernet --with-sdcard

with no error messages. If *that* doesn't work for you, I'd
double-check the LiteX, riscv64-unknown-linux-gnu-* toolchain, and
Vivado installations for problems.

3. Rocket variants are getting out of hand, and I'm in the process of
completely reworking them, with "linux" actually having an FPU, and
"full" including hypervisor support on top, and core-count & memory
bus width being additional parameters on the build command line. This
should happen within a week or two, so until then everything I said
in 1) and 2) above should still be valid.

HTH,
--Gabriel

gsomlo, 2023-03-16:
Oh and one more thing. I built my own riscv64-unknown-linux-gnu-*
toolchain from the sources at https://github.com/riscv/riscv-gnu-toolchain

It's a few months old now, so probably due for an update, but still
works perfectly fine for me. Grab the pre-built binaries from here:

http://www.contrib.andrew.cmu.edu/~somlo/BTCP/RISCV-toolchain.tar.xz

or follow the instructions at
https://github.com/litex-hub/linux-on-litex-rocket to build your own
from source.

n-kremeris, 2023-03-16:
Sorry, i forgot to confirm that the workaround provided by @AEW2015 fixes my issue. Just want to raise the issue that it's broken by default on Ubuntu if one follows the "getting started" guide.

shenki, 2023-03-22:
I tried this command line:

>  ./litex-boards/litex_boards/targets/digilent_nexys_video.py --build --cpu-type rocket --cpu-variant linuxd --sys-clk-freq 50e6 --with-sdcard

and was unable to reproduce the error. I was using Ubuntu 22.02 with riscv64-unknown-elf-gcc 10.2.0. However, if I install the linux toolchain it does reproduce.

Which getting started guide are you following? We should update it to ensure the elf toolchain is recommended, instead of the Linux one.

jwfaye, 2023-04-07:
Hello, I also have had this issue since I upgraded to Ubuntu 22.04. I tried several toolchains but still. When I compile a code with floats, I have these errors from the compiler : 

Adding "CFLAGS += -march=rv32imac -mabi=ilp32" to makefile fixes the error, but there is no output when I launch the litex_term command. 
What is weird is that on Ubuntu20.04, it works fine by changing the toolchain with the one provided by @AEW2015.

jwfaye, 2023-04-07:
Hello, Back again. Using the old version of Litex (04/2022 release) and the toolchain given by @AEW2015 worked for me.

JamesTimothyMeech, 2023-06-06:
I tried a simple example with the default toolchain that litex that litex points to in variables.mak. This is my result without changing anything in variables.mac:

`meechy@DESKTOP-5HD2OJR:~/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add$ make
 CC       crt0.o
echo "Compiling"
Compiling
 CC       main.o
main.c:20:10: warning: no previous prototype for 'pcg32_random_r' [-Wmissing-prototypes]
   20 | uint32_t pcg32_random_r(pcg32_random_t* rng)
      |          ^~~~~~~~~~~~~~
main.c:31:7: warning: no previous prototype for 'uint_to_float' [-Wmissing-prototypes]
   31 | float uint_to_float(uint32_t x)
      |       ^~~~~~~~~~~~~
main.c:37:1: warning: no previous prototype for 'add' [-Wmissing-prototypes]
   37 | add(float a, float b) {
      | ^~~
main.c:42:1: warning: no previous prototype for 'gsl_ran_gaussian' [-Wmissing-prototypes]
   42 | gsl_ran_gaussian (const float sigma)
      | ^~~~~~~~~~~~~~~~
main.c:149:13: warning: 'console_service' defined but not used [-Wunused-function]
  149 | static void console_service(void)
      |             ^~~~~~~~~~~~~~~
/home/meechy/Casino/Enjoy_Digital/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-gcc -nostdlib -nodefaultlibs -Wl,--no-dynamic-linker -Wl,--build-id=none -MD -MP -Os -march=rv32i2p0_mc   -mabi=ilp32 -D__femtorv__  -g3 -no-pie -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libbase -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/include -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software -I/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include -I/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include/../libc -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/cores/cpu/femtorv -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -I/home/meechy/Casino/gsl-2.7.1 -L/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include -L/home/meechy/Casino/gsl-2.7.1/.libs -T linker.ld -N -o demo.elf \
        crt0.o main.o \
        -L../build/digilent_arty/software/libc -L../build/digilent_arty/software/libcompiler_rt -L../build/digilent_arty/software/libbase -L../build/digilent_arty/software/libfatfs -L../build/digilent_arty/software/liblitespi -L../build/digilent_arty/software/liblitedram -L../build/digilent_arty/software/libliteeth -L../build/digilent_arty/software/liblitesdcard -L../build/digilent_arty/software/liblitesata -L../build/digilent_arty/software/bios \
        -Wl,--whole-archive \
        -Wl,--gc-sections \
        -Wl,-Map,demo.elf.map \
        -lc -lcompiler_rt -lbase -lfatfs -llitespi -llitedram -lliteeth -llitesdcard -llitesata
main.o: In function `gsl_ran_gaussian':
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:50: undefined reference to `__extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to `__extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to `__truncdfsf2'
main.o: In function `main':
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:184: undefined reference to `__extendsfdf2'
collect2: error: ld returned 1 exit status
make: *** [Makefile:20: demo.elf] Error 1`

This is my result when I change the `TRIPLE` variable in variables.mak to point to the new toolchain at: `TRIPLE=/home/meechy/Casino/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf`

The new toolchain still produces errors for me: 
`meechy@DESKTOP-5HD2OJR:~/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add$ make
 CC       crt0.o
echo "Compiling"
Compiling
 CC       main.o
main.c:20:10: warning: no previous prototype for 'pcg32_random_r' [-Wmissing-prototypes]
   20 | uint32_t pcg32_random_r(pcg32_random_t* rng)
      |          ^~~~~~~~~~~~~~
main.c:31:7: warning: no previous prototype for 'uint_to_float' [-Wmissing-prototypes]
   31 | float uint_to_float(uint32_t x)
      |       ^~~~~~~~~~~~~
main.c:37:1: warning: no previous prototype for 'add' [-Wmissing-prototypes]
   37 | add(float a, float b) {
      | ^~~
main.c:42:1: warning: no previous prototype for 'gsl_ran_gaussian' [-Wmissing-prototypes]
   42 | gsl_ran_gaussian (const float sigma)
      | ^~~~~~~~~~~~~~~~
main.c:149:13: warning: 'console_service' defined but not used [-Wunused-function]
  149 | static void console_service(void)
      |             ^~~~~~~~~~~~~~~
/home/meechy/Casino/Enjoy_Digital/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-gcc -nostdlib -nodefaultlibs -Wl,--no-dynamic-linker -Wl,--build-id=none -MD -MP -Os -march=rv32i2p0_mc   -mabi=ilp32 -D__femtorv__  -g3 -no-pie -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libbase -I/home/meechy/Cas [... cut ...]
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:50: undefined reference to `__extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to `__extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to `__truncdfsf2'
main.o: In function `main':
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:184: undefined reference to `__extendsfdf2'
collect2: error: ld returned 1 exit status
make: *** [Makefile:20: demo.elf] Error 1`

JamesTimothyMeech, 2023-06-06:
Adding CFLAGS += -march=rv32imac -mabi=ilp32  to variables.mak does not appear to change anything for me:
`meechy@DESKTOP-5HD2OJR:~/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add$ make
 CC       crt0.o
echo "Compiling"
Compiling
 CC       main.o
main.c:20:10: warning: no previous prototype for 'pcg32_random_r' [-Wmissing-prototypes]
   20 | uint32_t pcg32_random_r(pcg32_random_t* rng)
      |          ^~~~~~~~~~~~~~
main.c:31:7: warning: no previous prototype for 'uint_to_float' [-Wmissing-prototypes]
   31 | float uint_to_float(uint32_t x)
      |       ^~~~~~~~~~~~~
main.c:37:1: warning: no previous prototype for 'add' [-Wmissing-prototypes]
   37 | add(float a, float b) {
      | ^~~
main.c:42:1: warning: no previous prototype for 'gsl_ran_gaussian' [-Wmissing-prototypes]
   42 | gsl_ran_gaussian (const float sigma)
      | ^~~~~~~~~~~~~~~~
main.c:149:13: warning: 'console_service' defined but not used [-Wunused-function]
  149 | static void console_service(void)
      |             ^~~~~~~~~~~~~~~
/home/meechy/Casino/Enjoy_Digital/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-gcc -nostdlib -nodefaultlibs -Wl,--no-dynamic-linker -Wl,--build-id=none -MD -MP -Os -march=rv32imac -mabi=ilp32  -g3 -no-pie -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libbase -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/include -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software -I/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include -I/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include/../libc -I/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/cores/cpu/femtorv -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -I/home/meechy/Casino/gsl-2.7.1 -L/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/software/include -L/home/meechy/Casino/gsl-2.7.1/.libs -T linker.ld -N -o demo.elf \
        crt0.o main.o \
        -L../build/digilent_arty/software/libc -L../build/digilent_arty/software/libcompiler_rt -L../build/digilent_arty/software/libbase -L../build/digilent_arty/software/libfatfs -L../build/digilent_arty/software/liblitespi -L../build/digilent_arty/software/liblitedram -L../build/digilent_arty/software/libliteeth -L../build/digilent_arty/software/liblitesdcard -L../build/digilent_arty/software/liblitesata -L../build/digilent_arty/software/bios \
        -Wl,--whole-archive \
        -Wl,--gc-sections \
        -Wl,-Map,demo.elf.map \
        -lc -lcompiler_rt -lbase -lfatfs -llitespi -llitedram -lliteeth -llitesdcard -llitesata
main.o: In function gsl_ran_gaussian':
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:50: undefined reference to __extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to __extendsfdf2'
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:55: undefined reference to __truncdfsf2'
main.o: In function main':
/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/float-add/main.c:184: undefined reference to __extendsfdf2'
collect2: error: ld returned 1 exit status
make: *** [Makefile:20: demo.elf] Error 1`

closed 2023-07-31 by enjoy-digital