enjoy-digital/litex #1630: Is there any tutorial about how to connect two components through wishbone or AXI bus?

limingxuan-pku, 2023-03-03:
I have go through the wiki, but it does not have an example about how to do that. So where and how can I learn to use the bus?

enjoy-digital, 2023-03-10:
Hi @limingxuan-pku,

I'm planning to write a wiki page with examples for this. For now, can you provide more information about your use-case? I could probably redirect you to some code. Are you for example wrapping two components with LiteX and trying to connect them? Or reusing existing LiteX component and just want to connect them?

For AXI/AXI-Lite, you can find some examples in:
https://github.com/enjoy-digital/litex_verilog_axi_test/blob/master/test_axi.py
https://github.com/enjoy-digital/litex_verilog_axi_test/blob/master/test_axi2axi_lite.py

limingxuan-pku, 2023-03-10:
> For now, can you provide more information about your use-case? I could probably redirect you to some code. Are you for example wrapping two components with LiteX and trying to connect them? Or reusing existing LiteX component and just want to connect them?
Hi @enjoy-digital! Thanks for your reply. My use-case is very simple right now: I just need to connect my own module written in verilog to CPU through wishbone, which means I'd like to add my module as the wishbone bus slave.

fjullien, 2023-03-10:
You may find some examples here: https://github.com/fjullien/migen_litex_tutorials

closed 2023-04-03 by limingxuan-pku