enjoy-digital/litex #1644: Sipeed tang nano 9k uart problem

mgaletakis, 2023-03-11:
Hello, 

I am trying to test litex on sipeed tang_nano_9k board. Litex is installed on WSL on windows and tested with various board like the digilent arty-7. The sipped board seems to be programmed and alive i.e the ledchaser works but nothing happens on the  serial port it is stacked. 
The board hardware is ok if i program it with gowin software with a  uart demo it works fine both in windows and WSL.

Did someone else had similar problems? Any hints ?
Thank you in advance 
M

trabucayre, 2023-03-11:
Your issue seems similar to [this one](https://github.com/litex-hub/litex-boards/issues/481).
Could you try proposed test?
Thanks.

mgaletakis, 2023-03-11:
Yes i will try it first think Monday morning (hardware is in my office) and maybe try also to move rom section in ram 
Thank you i will get back with results soon
M

trabucayre, 2023-03-11:
I have modified target to remove 
```
kwargs["integrated_rom_size"] = 0
```
and
```
self.bus.add_region("rom", SoCRegion(
    origin = self.bus.regions["spiflash"].origin + bios_flash_offset,
    size   = 64*kB,
    linker = True)
)
self.cpu.set_reset_address(self.bus.regions["rom"].origin)
```

it's confirm my assumption:

```
ERROR (RP0002) : The number(31) of BSRAM in the design exceeds the resource limit(26) of current device. And RAM_STYLE maybe the useful user assignment to change the inference result
```
This why the rom must be stored into the flash.

mgaletakis, 2023-03-13:
Hi @trabucayre 
The board is working fine if you use flash form bios. I am using WSL on windows and openFPGAloader.  Also, as mentioned in the other issue you pointed, the flashing procedure is slow.
[image]

Thank you for your support i will close this issue if you agree.

trabucayre, 2023-03-13:
Thanks for your feedback!
Yes, as mentionned in the other issue with litlebee devices configured in on-chip mode the SPI flash access must be done in *JTAG Boundary Scan Mode* (see [UG290E](http://cdn.gowinsemi.com.cn/UG290E.pdf p.58).
In this mode the SPI flash is bitbanged an consecuently time to program is awfully slow.

You can close it yep :)

closed 2023-03-13 by mgaletakis

luxuan12512, 2024-01-05:
> 合作
hello,i have quetion about it .
i dele the del the code 

kwargs["integrated_rom_size"] = 0

and

self.bus.add_region("rom", SoCRegion(
    origin = self.bus.regions["spiflash"].origin + bios_flash_offset,
    size   = 64*kB,
    linker = True)
)
self.cpu.set_reset_address(self.bus.regions["rom"].origin)

and run in terminal:
./sipeed_tang_nano_9k.py --build
is no problem,and uart is out 
but run in termimal:
./sipeed_tang_nano_9k.py  --with-spi-sdcard   --build
have error 
ERROR (RP0002) : The number(31) of BSRAM in the design exceeds the resource limit(26) of current device. And RAM_STYLE maybe the useful user assignment to change the inference result
 
if i don't del the code to run :
./sipeed_tang_nano_9k.py  --with-spi-sdcard   --build
is no error but  uart isn't out 

So what should I do to be able to have urat output in the --with-spi-sdcard 
thanks!!!

Jiraphat-DEV, 2024-03-31:
Hello, I have the same problem as you. And I read and still don't understand how to fix this problem. Can you please explain it to me? Thank you very much