enjoy-digital/litex #1645: liblitesdcard: SD Card write speed issue

bayi, 2023-03-12:
Hi,

TLDR:  CMD12 should not be sent out after single block writes to the SDCARD, this causes degraded write performance.

I was debugging the reason why in my modiied litex bios where i have write support enabled in libfatfs my mkdir commands took over 30 seconds. After a many hours of debugging i found this in `liblitesdcard`: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitesdcard/sdcard.c#L555

According to the SDCARD/SDIO datasheets i found (and also the comment above the linked line) `sdcard_stop_transmission()` which sends out CMD12 ( aka "STOP_TRANSMISSION") should only be called after a multi block transfer, otherwise it timeouts but it still works fine, as fatfs is calling multiple single block writes on mkdir and rm while writing data was fine as it is using multi block writes there.

After adding a
```c
        if (nblocks > 1)
            sdcard_stop_transmission();
```
At line 554-555

If guard (the same thats also in the `disk_read()` function) the write speeds drastically improved ( mkdir is now taking only a couple ms instead of 30+ seconds ). 

Im still testing this but there are no side effects/corruption so far from this. 

Maybe this issue will help somebody in the future, also if you want i can send in a PR for this.

PS.: Thanks for the work you put into this, hope i can contribute more in the future.

enjoy-digital, 2023-03-13:
Thanks @bayi, I'll have closer look very soon.

bayi, 2023-03-13:
Steps to reproduce:

1. In `soc/software/libfatfs/ffconf.h` change `FF_FS_READONLY` to 0
2. in `soc/software/liblitesdcard/sdcard.c` add `sd_disk_write` to `SdcardDiskops.disk_write` also add an empty function which returns FR_OK as `.disk_ioctl` othwerise libfatfs will do a zeroptr write ;)
3. Add some bios/cmds like mkdir, rm etc. they just simply need to call the libfatfs functions (`f_mkdir` , `f_unlink` )

Now try it with and without the before mentioned modification, especially mkdir is interesting because it took me too lonk.

Since then i already made some tests ( and benchmarks ) but im stuck in work rn, will post them later if you want. So far no filesystem corruption i tested it with 2 different sdcard models (Samsung and Sandisk)

bayi, 2023-03-13:
Here is my disk_ioctl implementation i dont know if it matters ( according to the documentation we are always doing sync writes so it shouldnt )
```c
static DRESULT disk_ioctl(BYTE pdrv, BYTE cmd, void* buff) 
{
    switch(cmd) {

        case CTRL_SYNC:
            while ((sdblock2mem_dma_done_read() & 0x1) == 0);
            while ((sdmem2block_dma_done_read() & 0x1) == 0);
        break;

        case GET_SECTOR_COUNT:
            *(DWORD*)buff=4096;
        break;

        case GET_SECTOR_SIZE:
            *(WORD* ) buff = 512;
        break;

        case GET_BLOCK_SIZE:
            *(WORD* ) buff = 32;
        break;

    }
    return RES_OK;
}
```

bayi, 2023-03-13:
Some test results for 10 directory creation that i just did:

# Without patch
mkdir avg: **24512ms**

# With patch
mkdir avg: **63ms**

Card: SanDisk microSDHC Ultra 32GB

(yes work is so boring ... ;)

enjoy-digital, 2023-03-24:
Thanks @bayi, this has been integrated with https://github.com/enjoy-digital/litex/commit/e55f0da7c71cc2d9833921b38cf1ed23c2dfe38b.

closed 2023-03-24 by enjoy-digital