enjoy-digital/litex #1661: Problem when adding a second sinal on LiteScope

KevinQhv, 2023-03-24:
Hi !

I am working on the digilent Arty A7-100T board with LiteX version 2022.08. I am using the UARTBone for communication with LiteScope. 

I want to observe the registers of the CVA6 RISC-V CPU. I use the LiteScope tool. I tried to display a register when a signal is triggered and it is successful. But, when I add another register, the LiteScope script stops at `[running] ...`. 
I use the command : `litescope_cli` or `litescope_cli -r main_trigger_signal` if i want the signal at the trigger moment.
When i just use one signal register, i am no problem. But no with 2 signals.

Added in the file target :

```     
        # GPIO Trigger 
        self.submodules.TRIGGER=GPIOOut(platform.request("tio_trigger"), 0)
        self.add_csr("TRIGGER")

        self.trigger_signal = Signal()

        self.comb += self.trigger_signal.eq(self.TRIGGER.out.storage)

        # UARTBone ----------------------------------------------------------------
        self.add_uartbone(name="serial_LiteScope", baudrate=115200)
        
        # LiteScope -------------------------------------------------------------------
        analyzer_signals = [
            self.cpu.reg_a5_LiteScope,
            self.cpu.reg_a2_LiteScope,
            self.trigger_signal
        ]

        self.submodules.analyzer = LiteScopeAnalyzer(analyzer_signals,
            depth        = 512,
            clock_domain = "sys",
            samplerate   = sys_clk_freq,
            csr_csv      = "analyzer.csv"
        )

```

Added the signals in cva6/core.py : 

```
def __init__(self, platform, variant="standard", convert_periph_bus_to_wishbone=True):
        self.platform     = platform
        self.variant      = variant
        self.reset        = Signal()
        self.interrupt    = Signal(32)
        # Peripheral bus (Connected to main SoC's bus).
        self.axi_if = axi_if = axi.AXIInterface(data_width=64, address_width=32, id_width=4)
        if convert_periph_bus_to_wishbone:
            self.wb_if = wishbone.Interface(data_width=axi_if.data_width,
                                            adr_width=axi_if.address_width - log2_int(axi_if.data_width // 8))
            self.submodules += axi.AXI2Wishbone(axi_if, self.wb_if)
            self.periph_buses = [self.wb_if]
        else:
            self.periph_buses = [axi_if]
        # Memory buses (Connected directly to LiteDRAM).
        self.memory_buses = []

        self.reg_a5_LiteScope =Signal(32) # Signal a5 register
        self.reg_a2_LiteScope =Signal(32) # Signal a2 register

        # CPU Instance.
        self.cpu_params = dict(
            # Clk / Rst.
            i_clk_i       = ClockSignal("sys"),
            i_rst_n       = ~ResetSignal("sys") | self.reset,

            o_registerLiteScopea5_o = self.reg_a5_LiteScope, # Signal a2 register
            o_registerLiteScopea2_o = self.reg_a2_LiteScope, # Signal a2 register

            # Interrupts
            i_irq_sources = self.interrupt,

            # AXI interface.
            o_AWVALID_o   = axi_if.aw.valid,
            i_AWREADY_i   = axi_if.aw.ready,
            o_AWID_o      = axi_if.aw.id,
            o_AWADDR_o    = axi_if.aw.addr,
            o_AWLEN_o     = axi_if.aw.len,
            o_AWSIZE_o    = axi_if.aw.size,
            o_AWBURST_o   = axi_if.aw.burst,
            o_AWLOCK_o    = axi_if.aw.lock,
            o_AWCACHE_o   = axi_if.aw.cache,
            o_AWPROT_o    = axi_if.aw.prot,
            o_AWQOS_o     = axi_if.aw.qos,
            o_AWREGION_o  = Open(),
            o_AWUSER_o    = Open(),

            o_WVALID_o    = axi_if.w.valid,
            i_WREADY_i    = axi_if.w.ready,
            o_WDATA_o     = axi_if.w.data,
            o_WSTRB_o     = axi_if.w.strb,
            o_WLAST_o     = axi_if.w.last,
            o_WUSER_o     = Open(),

            i_BVALID_i    = axi_if.b.valid,
            o_BREADY_o    = axi_if.b.ready,
            i_BID_i       = axi_if.b.id,
            i_BRESP_i     = axi_if.b.resp,
            i_BUSER_i     = 0,

            o_ARVALID_o   = axi_if.ar.valid,
            i_ARREADY_i   = axi_if.ar.ready,
            o_ARID_o      = axi_if.ar.id,
            o_ARADDR_o    = axi_if.ar.addr,
            o_ARLEN_o     = axi_if.ar.len,
            o_ARSIZE_o    = axi_if.ar.size,
            o_ARBURST_o   = axi_if.ar.burst,
            o_ARLOCK_o    = axi_if.ar.lock,
            o_ARCACHE_o   = axi_if.ar.cache,
            o_ARPROT_o    = axi_if.ar.prot,
            o_ARQOS_o     = axi_if.ar.qos,
            o_ARUSER_o    = Open(),
            o_ARREGION_o  = Open(),

            i_RVALID_i    = axi_if.r.valid,
            o_RREADY_o    = axi_if.r.ready,
            i_RID_i       = axi_if.r.id,
            i_RDATA_i     = axi_if.r.data,
            i_RRESP_i     = axi_if.r.resp,
            i_RLAST_i     = axi_if.r.last,
            i_RUSER_i     = 0,
        )

```

enjoy-digital, 2023-03-27:
Hi @KevinQhv,

are the timing passing? (CVA6 probably requires using `--sys-clkj-freq` parameter of the target to lower the frequency) You'll also maybe want to use `register=True` on the LiteScope instance to improve them.

enjoy-digital, 2023-03-28:
> The clock is working well. It's not the problem. 
I'm just worried about the place & route timings, what is your `--sys-clk-freq `and are the timing met with Vivado?

closed 2023-12-08 by KevinQhv