enjoy-digital/litex #1664: How used FreeRTOS with LiteX ?

Marcelin000000, 2023-03-28:
Hi !

I want to implement RTOS in my SoC built with LiteX. I want to use FreeRTOS on the CVA6 CPU. But, I don't see how to link FreeRTOS to my SoC `csr.h` file.

enjoy-digital, 2023-04-04:
Hi @Marcelin000000,

we don't yet have FreeRTOS support in LiteX, but you can probably use ChatGPT to simplify adding FreeRTOS support. 
Here is an outline of the steps you could follow:

[image]
[image]

Feel free to also play yourself with ChatGPT to get more indications (ChatGPT has a general understanding of both FreeRTOS and LiteX, it will not create the port for you, but can help you figure our the different steps involved).

LukeGary462, 2025-08-04:
commenting here since I just was trying to answer the same question. It appears that neorv32 has a funtional FreeRTOS port. I have yet to try it myself, but that seems to be the simplest starting point without respect to platform specific peripherals.