enjoy-digital/litex #1667: How to implement DMA as a master

limingxuan-pku, 2023-04-03:
I'm new to SoC interconnection and not familiar with it. Now, I'd like to transfer data by using DMA.
I use class `WishboneDMAReader` & `WishboneDMAWriter` in `litex/soc/cores/dma.py` to implement my own DMA as following.
```
class my_dma(Module, AutoCSR):
    def __init__(self):
        self.bus = wishbone.Interface()
        
        self.submodules.reader = WishboneDMAReader(bus=self.bus, with_csr=True)
        self.submodules.writer = WishboneDMAWriter(bus=self.bus, with_csr=True)
        
        self.comb += self.reader.source.connect(self.writer.sink)
        self.comb += self.reader.fifo.sink.ready.eq(1)
```
Then, I add `my_dma` as a master:
```
        self.submodules.my_dma = my_dma()
        self.bus.add_master(name="my_dma", master=self.my_dma.bus)
```
After that, I try to use CSRs to indicate the address, length, etc. However, I can't successfully transfer data from one address to another, for example, moving data at `0x7000_0000` to `0x9500_0000`. I think there is something wrong with `my_dma` class, but I'm not able to figure it out.

enjoy-digital, 2023-04-03:
Hi @limingxuan-pku,

The module seems almost correct: `self.comb += self.reader.fifo.sink.ready.eq(1)` should not be required.

Can you share the C code you are using?

enjoy-digital, 2023-04-03:
Here is also an answer and some pseudo code from ChatGPT that seems correct to me:
[image]

limingxuan-pku, 2023-04-04:
> Can you share the C code you are using?

Hi @enjoy-digital !
According to `csr.json` generated by the script, I use pointer in C code to configure CSRs.
Here is the generated `csr.json`:
```
...
        "my_dma_reader_base": {
            "addr": 4026542080,
            "size": 2,
            "type": "rw"
        },
        "my_dma_reader_length": {
            "addr": 4026542088,
            "size": 1,
            "type": "rw"
        },
        "my_dma_reader_enable": {
            "addr": 4026542092,
            "size": 1,
            "type": "rw"
        },
        "my_dma_reader_done": {
            "addr": 4026542096,
            "size": 1,
            "type": "ro"
        },
        "my_dma_reader_loop": {
            "addr": 4026542100,
            "size": 1,
            "type": "rw"
        },
        "my_dma_reader_offset": {
            "addr": 4026542104,
            "size": 1,
            "type": "ro"
        },
        "my_dma_writer_base": {
            "addr": 4026542108,
            "size": 2,
            "type": "rw"
        },
        "my_dma_writer_length": {
            "addr": 4026542116,
            "size": 1,
            "type": "rw"
        },
        "my_dma_writer_enable": {
            "addr": 4026542120,
            "size": 1,
            "type": "rw"
        },
        "my_dma_writer_done": {
            "addr": 4026542124,
            "size": 1,
            "type": "ro"
        },
        "my_dma_writer_loop": {
            "addr": 4026542128,
            "size": 1,
            "type": "rw"
        },
        "my_dma_writer_offset": {
            "addr": 4026542132,
            "size": 1,
            "type": "ro"
        },
...
```
Here is my C code:
```
    uint32_t* p_dma_csr_r = (uint32_t*)4026542080;
    uint32_t* p_dma_csr_w = (uint32_t*)4026542108;
    uint32_t* p_conv = (uint32_t*)0x95000000;
    uint32_t* p_ram = (uint32_t*)0x70000000;
    for ( int i = 0; i < 16; i++ )
        printf("Data at address %p = %08lx\n", p_dma_csr_r+i, *(p_dma_csr_r+i));
    *p_dma_csr_r = 0x70000000;
    *(p_dma_csr_r+2) = 4;
    *(p_dma_csr_r+3) = 1;
    *p_dma_csr_w = 0x95000008;
    *(p_dma_csr_w+2) = 4;
    *(p_dma_csr_w+3) = 1;
    printf("\n");
    for ( int i = 0; i < 16; i++ )
        printf("Data at address %p = %08lx\n", p_dma_csr_r+i, *(p_dma_csr_r+i));
    printf("\n");
    printf("Data at address %p = %08lx\n", p_ram, *p_ram);
    for ( int i = 0; i < 4; i++ )
        printf("Data at address %p = %08lx\n", p_conv+i, *(p_conv+i));
```
And finally, the outcome of my C code:
```
Data at address 0xf0002800 = 00000000
Data at address 0xf0002804 = 00000000
Data at address 0xf0002808 = 00000000
Data at address 0xf000280c = 00000000
Data at address 0xf0002810 = 00000000
Data at address 0xf0002814 = 00000000
Data at address 0xf0002818 = 00000000
Data at address 0xf000281c = 00000000
Data at address 0xf0002820 = 00000000
Data at address 0xf0002824 = 00000000
Data at address 0xf0002828 = 00000000
Data at address 0xf000282c = 00000000
Data at address 0xf0002830 = 00000000
Data at address 0xf0002834 = 00000000
Data at address 0xf0002838 = 00000000
Data at address 0xf000283c = 00000000

Data at address 0xf0002800 = 70000000
Data at address 0xf0002804 = 00000000
Data at address 0xf0002808 = 00000004
Data at address 0xf000280c = 00000001
Data at address 0xf0002810 = 00000000
Data at address 0xf0002814 = 00000000
Data at address 0xf0002818 = 00000000
Data at address 0xf000281c = 95000008
Data at address 0xf0002820 = 00000000
Data at address 0xf0002824 = 00000004
Data at address 0xf0002828 = 00000001
Data at address 0xf000282c = 00000000
Data at address 0xf0002830 = 00000000
Data at address 0xf0002834 = 00000000
Data at address 0xf0002838 = 00000000
Data at address 0xf000283c = 00000000

Data at address 0x70000000 = 11111111
Data at address 0x95000000 = 00000000
Data at address 0x95000004 = 00000000
Data at address 0x95000008 = 00000000
Data at address 0x9500000c = 00000000
```

enjoy-digital, 2023-04-04:
Hi @limingxuan-pku,

LiteX also generate a `csr.h `and I would recommend using this (at least as a start) to make your code a lot more readable: Each CSR has its access function that you can directly use.

limingxuan-pku, 2023-04-05:
> LiteX also generate a `csr.h `and I would recommend using this (at least as a start) to make your code a lot more readable: Each CSR has its access function that you can directly use.

Your suggestion works very well to me! Thanks! I've successfully moved data by using the function in `csr.h`.

limingxuan-pku, 2023-04-09:
Hi!
I've tried to using DMA to move data from my own slave to one variable in my C code whose address is located at the default main ram.
Here is my DMA module:
```
        self.submodules.my_dma_r = WishboneDMAReader(bus=self.bus_dma_r, with_csr=True)
        self.submodules.my_dma_w = WishboneDMAWriter(bus=self.bus_dma_w, with_csr=True)
        self.bus.add_master(name="my_dma_r", master=self.my_dma_r.bus)
        self.bus.add_master(name="my_dma_w", master=self.my_dma_w.bus)
        self.comb += self.my_dma_w.sink.data.eq(self.my_dma_r.source.data)
```
Here is my C code:
```
    uint32_t a = 0;
    printf("Data at %p = %08lx\n", &a, a);

    my_dma_r_base_write(0x70000000);
    my_dma_r_length_write(4);
    my_dma_r_enable_write(1);
    my_dma_w_base_write((uint64_t)&a);
    my_dma_w_length_write(4);
    my_dma_w_enable_write(1);

    printf("Data at %p = %08lx\n", &a, a);
```
What I do is to move data at 0x7000_0000 (which should be 0x1111_1111) to variable `a`.
However, value `a` always equals to 0.

offnaria, 2023-04-28:
Hi @limingxuan-pku 

I think the problem is that you connected only the `data` signals, but it should also need `valid`, `ready` signals.
I've not verified but could you try using:
```
self.comb += self.my_dma_r.source.connect(self.writer.sink)
```
instead of:
```
self.comb += self.my_dma_w.sink.data.eq(self.my_dma_r.source.data)
```

Beauxrel, 2023-06-26:
Can someone explain to me how this works? I want to add dma so that my fmc module can access memory. Do you add the my_dma class to the dma.py file? Can you tell me all the files I need to change to reproduce this.

Thank you!

bentwire, 2023-09-14:
It seems like you actually have to start the writer first.  Otherwise the reader reads until its done and the writer never starts.

I used litex_cli to test manually.

bentwire, 2023-09-18:
So I wrote this: https://gist.github.com/bentwire/66d08e72540c4f52e39f58c1c41f1124
Based on some of the examples here and the DMA code itself.  

It seems to work, but I have a few questions:

1. Why do I need to create a stream when setting up the writer FSM?  I copied this from the Wishbone DMA code but I don't quite understand it.  Is it to give access to the address portion of the stream to set the offset?  Can that not be done directly on the stream from the writer itself?
2. I want to build a MEM->WB and WB->MEM version but I will need to adapt the SDRAM bus width and address to the wishbone address and width...  How do I do this with streams, is there some sort of adapter?
3. My code is limited to 32bit xfers on 32bit boundaries from how I understand the underlying DMA primitives.  Is this true?

Thanks for any help!