enjoy-digital/litex #1668: No output after "Liftoff!" on NexysA7/CVA6

djexvr, 2023-04-06:
Hello,
I am trying to implement a cva6 on a Nexys A7 board. When I load the image on the board and try to load the demo through UART the program get stuck after this output
```
 __   _ __  _  _m
   // / -_)>[0m
m   ___/_/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 13 2023 18:17:56
 BIOS CRC passed (73e6)

 LiteX git sha1: d256a5e3

--=============== SoC ==================--
CPU:		CVA6 @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
MAIN-RAM:	64.0KiB

--========== Initialization ============--
Memtest at 0x40000000 (64.0KiB)...
  Write: 0x40000004001 64.0KiB      
mtesRead: 0x4004001 64.   
Meed a4000 (Setial.0Ki.
 te s: 176918GiB/s
   Repeed16.10s

========= Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x40000000 (5552 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```
(I have some minor display problems causing characters to be mixed up but I don't know where they're coming from.)

To build the image I used this command `./digilent_nexys4ddr.py --build --load --cpu-type cva6 --sys-clk-freq 50e6 --integrated-main-ram-size=0x10000`. If I don't use the `--integrated-ram-size` option The memtest during initialization fails and then it stops at "Liftoff".

I also tried running the vexriscv cpu on the same board and the demo worked well.

I tried using a Genesys 2 board for the cva6 but I got the same behavior.

Can anyone help me fix this issue ?

Thanks,

enjoy-digital, 2023-04-07:
Hi @djexvr,

the behavior and serial issues could be related to timings are that not satisfied. I would first recommend testing in simulation:

```
litex_sim --integrated-main-ram-size=0x10000 --cpu-type=vexriscv --no-compile-gateware
litex_bare_metal_demo --build-path=build/sim/
litex_sim --integrated-main-ram-size=0x10000 --cpu-type=cva6 --ram-init=demo.bin
```

If this works, have a closer look at the timing report generated with Vivado and maybe reduce `sys_clk_freq` to 25MHz.

djexvr, 2023-05-04:
Hello again,
I've made some progress in identifying the origin of the problem. First of all the simulation did work but modifying the `sys_clk_freq` did not solve the issue. I've tried to modify the bios to get more info on the bug. I've been able to verify that the program is correctly loaded in the ram. The problem seems to be located in the `uart_sync` call right after the Liftoff since no printf are executed after it is called. I am  not exactly sure on how it is supposed to work but it seems that `tx_consume` and `tx_produce` are never equal triggering an infinite loop.

OrkunAliOzkan, 2023-08-09:
Hi @djexvr, were you able to resolve this problem? I am encountering a similar situation with an SoC which I generate.

When attempting to boot busybox on `~/litex_environment/litex-boards/litex_boards/targets/digilent_nexys_video.py --build --cpu-type rocket --cpu-variant full --cpu-num-cores 1 --cpu-mem-width 2 --sys-clk-freq 50e6 --with-ethernet --with-sdcard --doc --csr-json ./csr.json`, the SoC fails at the same stage due to the same blocking reasons, where `tx_consume` and `tx_produce` equate

lianakoleva, 2023-10-04:
Also facing similar issue, opened #1794 with more details

roby2014, 2024-04-20:
Having a similar issue #1935

cuongnd537, 2025-11-19:
HI @djexvr, @lianakoleva, @OrkunAliOzkan and @enjoy-digital,
Did you resolve this CVA6 issue?
Currently, I'm facing the same problem and I've realized that it have been discussed in many threads, but it hasn't been completely solved yet.
If you have fixed it, please let me know what you did.
Thank you so much.