enjoy-digital/litex #1669: OSError: Error occured during Vivado's script execution.

feyzaozdemir, 2023-04-07:
Hi,
I wanna build an SoC with kcu105 board. I get some path errors and solved it. Then I get this error: "OSError: Error occured during Vivado's script execution." 
I saw #1343 . His path errors was same with me and I solved with same way. I see  @enjoy-digital s suggestion for OSError but I am using Vivado ML Enterprise Edition with license. I searched but couldn't find anything. What is causing my OSError? Any kind of help accepted.
Thanks.

trabucayre, 2023-04-07:
could you try : 

- first export `LITEX_ENV_VIVADO` or source settings64.sh
- and after running python script

like that
```
export LITEX_ENV_VIVADO=/somewhere/Vivado/$VERSION/
./xilinx_kcu105.py --build
```

where:
- `/somewhere` is the install dir
- `$VERSION` is something like 2019.2.

enjoy-digital, 2024-01-12:
Closing since answered and no feedback.

closed 2024-01-12 by enjoy-digital