enjoy-digital/litex #1678: Litex install is failing

pratipm, 2023-04-27:
Trying to install a fresh copy of litex is failing:
./litex_setup.py --init --install --user --config=full

........

[ 111.121] Installing liteiclink Git repository...
Obtaining file:///home/pratipm/litex/liteiclink
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      running egg_info
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/home/pratipm/litex/liteiclink/setup.py", line 11, in <module>
          setup(
        File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 108, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/usr/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/usr/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3.10/site-packages/setuptools/dist.py", line 1221, in run_command
          super().run_command(command)
        File "/usr/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.ensure_finalized()
        File "/usr/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
          self.finalize_options()
        File "/usr/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 253, in finalize_options
          if pd is not None and pd.key == self.egg_name.lower():
      AttributeError: 'PathDistribution' object has no attribute 'key'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Traceback (most recent call last):
  File "/home/pratipm/litex/./litex_setup.py", line 480, in <module>
    main()
  File "/home/pratipm/litex/./litex_setup.py", line 460, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
  File "/home/pratipm/litex/./litex_setup.py", line 285, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pip install --editable . {options}".format(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/usr/bin/python3" -m pip install --editable . --user' returned non-zero exit status 1.

OJthe123, 2023-05-05:
try to install pip3
sudo apt install python3-pip

CarlFK, 2023-05-22:
testing
 ./litex_setup.py --init --install --user --config=full

using  
' https://github.com/bjonnh/fpga_class_psone/blob/main/setup_linux.sh'

with this change
```
# python3 litex_setup.py --init
# python3 litex_setup.py --install
python3 ./litex_setup.py --init --install --user --config=full
```
the install does not error (good)
but the test fails (maybe bad?)

Install:
```
...
[ 265.251] Installing pythondata-cpu-vexriscv-smp Git repository...
Obtaining file:///home/videoteam/oss-cad-suite/litex/pythondata-cpu-vexriscv-smp
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Building wheels for collected packages: pythondata-cpu-vexriscv-smp
  Building editable for pythondata-cpu-vexriscv-smp (pyproject.toml) ... done
  Created wheel for pythondata-cpu-vexriscv-smp: filename=pythondata_cpu_vexriscv_smp-1.0.1.post325-0.editable-py3-none-any.whl size=3790 sha256=f277b933065ac51564bf87beb3199f4cd8351f778b7fb6136e665c2d565aea79
  Stored in directory: /tmp/pip-ephem-wheel-cache-bo107f3f/wheels/fb/db/50/d5adc2a731c1c1dd5dd97f46630cc622c06332c96de6686994
Successfully built pythondata-cpu-vexriscv-smp
Installing collected packages: pythondata-cpu-vexriscv-smp
Successfully installed pythondata-cpu-vexriscv-smp-1.0.1.post325
[ 268.931] Make sure that ~/.local/bin is in your PATH
[ 268.931] export PATH=$PATH:~/.local/bin # temporary (limited to the current terminal)
[ 268.931] or add the previous line into your ~/.bashrc to permanently update PATH
Everything is installed we can now test
```
test:
```
Cloning into 'alscope'...
remote: Enumerating objects: 22, done.
remote: Counting objects: 100% (22/22), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 22 (delta 8), reused 19 (delta 5), pack-reused 0
Receiving objects: 100% (22/22), 5.63 KiB | 1.88 MiB/s, done.
Resolving deltas: 100% (8/8), done.
Traceback (most recent call last):
  File "./main.py", line 10, in <module>
    from liteeth.phy.ecp5rgmii import LiteEthPHYRGMII
ModuleNotFoundError: No module named 'liteeth'
```

cklarhorst, 2023-05-23:
Hm, `liteeth` should have been installed with `python3 ./litex_setup.py --init --install --user --config=full`.
Maybe check the output of `python3 --version` and `pip --version`. 
They should both report the same python version. Otherwise, pip will install things into the wrong location.
(In my case the output is: `pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)` and `Python 3.10.6`

pcotret, 2023-06-08:
Got a similar issue, i had to `sudo <command>` in order to make it run :neutral_face: