enjoy-digital/litex #1694: LiteX attempts to use the wrong bin file when synthesizing FemtoRV petitbateau

JamesTimothyMeech, 2023-05-18:
When I run the command: ''

It seems like yosys is looking at the wrong chipdb file: `ERROR: Unable to read chipdb `/usr/share/nextpnr/xilinx-chipdb/xc7a35t.bin`.
It should be looking for `xc7a100t.bin`

Full error trace:

`Warnings: 67 unique messages, 67 total
End of script. Logfile hash: ff61b479a8, CPU: user 94.62s system 2.33s, MEM: 1067.12 MB peak
Yosys 0.28+6 (git sha1 cee3cb31b, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 28% 1x abc9_exe (38 sec), 24% 12x techmap (32 sec), ...
ERROR: Unable to read chipdb /usr/share/nextpnr/xilinx-chipdb/xc7a35t.bin
0 warnings, 1 error
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 221, in <module>
    main()
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 210, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/builder.py", line 367, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/soc.py", line 1322, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/platform.py", line 85, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/yosys_nextpnr.py", line 132, in build
    return YosysNextPNRToolchain.build(self, platform, fragment, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/yosys_nextpnr_toolchain.py", line 126, in build
    return GenericToolchain.build(self, platform, fragment, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/generic_toolchain.py", line 118, in build
    self.run_script(script)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/yosys_nextpnr_toolchain.py", line 232, in run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.`

trabucayre, 2023-05-18:
Good catch!
Could you update LiteX repo.
Fixed by [this commit](https://github.com/enjoy-digital/litex/commit/60537fc39f3efa9b5c222f15f469746b6ed87cb3)

JamesTimothyMeech, 2023-05-18:
Thanks for the fix, Is there an easy way to point LiteX to my version of chipdb? It currently looks in the wrong place

trabucayre, 2023-05-18:
Currently it's hardcoded and point to `/usr/share/nextpnr/xilinx-chipdb` but it's true it make sense to allows the use of an shell variable.

JamesTimothyMeech, 2023-05-18:
Thanks, I manually edited the file to point it to the correct place for me

closed 2023-05-18 by JamesTimothyMeech