enjoy-digital/litex #1695: Synthesizing FemtoRV petitbateu for the Arty 100T fails in two different ways for two different toolchains

JamesTimothyMeech, 2023-05-18:
I am trying to build FemtoRV petitbateu but it fails to build two different ways for two different toolchains.

When I build with yosys+nextpnr using command: `python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant petitbateau --variant a7-100 --toolchain yosys+nextpnr --load --build`

I get this error: `Info: Packing DRAM..
Info:    Transformed 0 tied-low DRAM address inputs to be tied-high
Info: Packing BRAM..
Info: Packing DSPs..
Info:     Created 6 DSP48E1_DSP48E1 cells from:
Info:              6x DSP48E1
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
build_digilent_arty.sh: line 4: 30381 Aborted                 nextpnr-xilinx --json digilent_arty.json --xdc digilent_arty.xdc --fasm digilent_arty.fasm --chipdb /home/meechy/Casino/nextpnr-xilinx/xilinx/xc7a100t.bin --write digilent_arty_routed.json --timing-allow-fail --seed 1
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 221, in <module>
    main()
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 210, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/builder.py", line 367, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/soc.py", line 1322, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/platform.py", line 85, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/yosys_nextpnr.py", line 132, in build
    return YosysNextPNRToolchain.build(self, platform, fragment, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/yosys_nextpnr_toolchain.py", line 126, in build
    return GenericToolchain.build(self, platform, fragment, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/generic_toolchain.py", line 118, in build
    self.run_script(script)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/yosys_nextpnr_toolchain.py", line 232, in run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.`

When I build with Vivado using command: `python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant petitbateau --variant a7-100 --toolchain vivado --load --build`

I get this error: `source digilent_arty.tcl
# create_project -force -name digilent_arty -part xc7a100tcsg324-1
# set_msg_config -id {Common 17-55} -new_severity {Warning}
# read_verilog {/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/femtorv32_petitbateau.v}
# read_verilog {/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v}
# read_verilog {/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/gateware/digilent_arty.v}
# read_xdc digilent_arty.xdc
# set_property PROCESSING_ORDER EARLY [get_files digilent_arty.xdc]
# synth_design -directive default -top digilent_arty -part xc7a100tcsg324-1 -include_dirs {/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX}
Command: synth_design -directive default -top digilent_arty -part xc7a100tcsg324-1 -include_dirs /home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a100t-csg324'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a100t-csg324'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 30773
ERROR: [Synth 8-2715] syntax error near + [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v:250]
ERROR: [Synth 8-2715] syntax error near + [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v:312]
ERROR: [Synth 8-2722] system function call clog2 is not allowed here [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v:156]
INFO: [Synth 8-2350] module PetitBateau ignored due to previous errors [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v:48]
INFO: [Synth 8-2350] module CLZ ignored due to previous errors [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/petitbateau.v:830]
ERROR: [Synth 8-988] EXECUTE_bit is already declared [/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/femtorv32_petitbateau.v:540]
Abnormal program termination (11)
Please check '/home/meechy/Casino/FPGA-System-on-Chip-Firmware/LiteX/build/digilent_arty/gateware/hs_err_pid30768.log' for details
Parent process (pid 30768) has died. This helper process will now exit
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 221, in <module>
    main()
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 210, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/builder.py", line 367, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/integration/soc.py", line 1322, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/platform.py", line 85, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/vivado.py", line 139, in build
    return GenericToolchain.build(self, platform, fragment, **kwargs)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/generic_toolchain.py", line 118, in build
    self.run_script(script)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/vivado.py", line 387, in run_script
    raise OSError("Error occured during Vivado's script execution.")
OSError: Error occured during Vivado's script execution.`

trabucayre, 2023-05-19:
I confirm:
- **petitbateau** fails with both *yosys_nextpnr* and *vivado* toolchains, but a `.bit` is produces by *f4pga* (I haven't tested in real hardware)
- **standard** variant fails with *yosys_nextpnr*, but a `.bit` is produces when using *vivado* and *f4pga* (also I haven't tested using an arty)

My first idea is to check if this situation is also true when using @BrunoLevy's *learn-fpga* and to compare *yosys* options used for *yosys_nextpnr* toolchain and by *f4pga* (latter has a dedicated tcl script shipped with the toolchain).

JamesTimothyMeech, 2023-05-19:
The learn-fpga version of petitbateau does build ok with yosys and nextpnr. In which file can I edit the options that LiteX uses when calling yosys and nextpnr? 

Unfortunately I have been struggling to properly install f4pga because it the last install instruction is difficult to understand and confirm that it has completed properly

JamesTimothyMeech, 2023-05-19:
@trabucayre is there an easy way to have my litex installation and my f4pga work at the same time? 

Currently LiteX does not work when the f4pga conda environment is activated: 
`(xc7) meechy@DESKTOP-5HD2OJR:~/Casino/FPGA-System-on-Chip-Firmware/LiteX$ python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant petitbateau --variant a7-100 --toolchain f4pga --load --build
/home/meechy/opt/f4pga/xc7/conda/envs/xc7/bin/python3: Error while finding module specification for 'litex_boards.targets.digilent_arty' (ModuleNotFoundError: No module named 'litex_boards')`

and f4pga does not work when I deactivate the conda environment: 
`meechy@DESKTOP-5HD2OJR:~/Casino/FPGA-System-on-Chip-Firmware/LiteX$ python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant petitbateau --variant a7-100 --toolchain f4pga --load --build
Traceback (most recent call last):
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/f4pga.py", line 19, in <module>
    from f4pga.flows.flow import Flow
ModuleNotFoundError: No module named 'f4pga'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 221, in <module>
    main()
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 184, in main
    soc = BaseSoC(
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 86, in __init__
    platform = digilent_arty.Platform(variant=variant, toolchain=toolchain)
  File "/home/meechy/Casino/Enjoy_Digital/litex-boards/litex_boards/platforms/digilent_arty.py", line 346, in __init__
    Xilinx7SeriesPlatform.__init__(self, device, _io, _connectors, toolchain=toolchain)
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/platform.py", line 40, in __init__
    from litex.build.xilinx import f4pga
  File "/home/meechy/Casino/Enjoy_Digital/litex/litex/build/xilinx/f4pga.py", line 25, in <module>
    raise ModuleNotFoundError("Try getting/updating F4PGA tool (https://github.com/chipsalliance/f4pga/)") from e
ModuleNotFoundError: Try getting/updating F4PGA tool (https://github.com/chipsalliance/f4pga/)`

I tried running `./litex_setup.py` with the conda environment activated and it completed without error but did not change anything.

JamesTimothyMeech, 2023-05-19:
Here are the options that work for petitbateau in learn-fpga to save you some time: https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/BOARDS/ulx3s.mk

`YOSYS_ULX3S_OPT=-DULX3S -q -p "synth_ecp5 -abc9 -top $(PROJECTNAME) -json $(PROJECTNAME).json"
NEXTPNR_ULX3S_OPT=--force --timing-allow-fail --json $(PROJECTNAME).json --lpf BOARDS/ulx3s.lpf \
                  --textcfg $(PROJECTNAME)_out.config --85k --freq 25 --package CABGA381

#######################################################################################################################

ULX3S: ULX3S.firmware_config ULX3S.synth ULX3S.prog

ULX3S.fast: ULX3S.synth ULX3S.prog_fast

ULX3S.synth: FIRMWARE/firmware.hex
        yosys $(YOSYS_ULX3S_OPT) $(VERILOGS)
        nextpnr-ecp5 $(NEXTPNR_ULX3S_OPT)
        ecppack --compress --svf-rowsize 100000 --svf $(PROJECTNAME).svf $(PROJECTNAME)_out.config $(PROJECTNAME).bit

ULX3S.show: FIRMWARE/firmware.hex
        yosys $(YOSYS_ULX3S_OPT) $(VERILOGS)
        nextpnr-ecp5 $(NEXTPNR_ULX3S_OPT) --gui

ULX3S.prog_fast: # program once (lost if device restarted)
        ujprog $(PROJECTNAME).bit

ULX3S.prog: # program permanently
        ujprog -j FLASH $(PROJECTNAME).bit

ULX3S.firmware_config:
        BOARD=ulx3s TOOLS/make_config.sh -DULX3S`

trabucayre, 2023-05-19:
These lines are used for *Lattice EPC5* devices, not for Xilinx. I have to find if *petitbateau* has been tested with *artix7*.

JamesTimothyMeech, 2023-05-19:
Ah that is a good point, my claim that the petitbateau from learn-fpga works for artix7 was based on that makefile completing so my original claim was not correct

trabucayre, 2023-05-19:
a `grep petitbateau` seems not showing anything with xilinx. I don't remember if @BrunoLevy has tested with an arty.

BrunoLevy, 2023-05-20:
Hi, I confirm that a while ago I was able to synthesize petitbateau on my arty A35T, using nextpnr-xilinx.

trabucayre, 2023-05-20:
@BrunoLevy: thanks for this information!

trabucayre, 2023-05-23:
With this fix *petibateau* work with `vivado` toolchain (tested with my arty A7 100T):

```
--- petitbateau.v   2023-05-05 18:10:11.048882475 +0200
+++ /tmp/petitbateau.v  2023-05-23 06:49:23.119789271 +0200
@@ -247,7 +247,7 @@
       // D' = denominator (rs2) normalized between [0.5,1] (set exp to 126)
       fpmi_gen(FPMI_FRCP_PROLOG); // D<-A; E<-B; A<-(-D'); B<-32/17; C<-48/17
       fpmi_gen_fma(0);            // X <- A*B+C (= -D'*32/17 + 48/17)
-      for(iter=0; iter<3; iter++) begin
+      for(iter=0; iter<3; iter=iter+1) begin
     if(PRECISE_DIV) begin
        // X <- X + X*(1-D'*X)
        // (slower more precise iter, but not IEEE754 compliant yet...)
@@ -309,7 +309,7 @@
       `FPMPROG_BEGIN(FPMPROG_SQRT);
       // D<-rs1; E,A,B<-(doom_magic - (A >> 1)); C<-3/2
       fpmi_gen(FPMI_FRSQRT_PROLOG);
-      for(iter=0; iter<2; iter++) begin
+      for(iter=0; iter<2; iter=iter+1) begin
     // X <- X * (3/2 - (0.5*rs1*X*X))           
     fpmi_gen(FPMI_LOAD_XY_MUL);  // X <- A*B; Y <- C
     fpmi_gen(FPMI_MV_A_X);       // A <- X

```

```
--- femtorv32_petitbateau.v 2023-05-05 18:10:11.048882475 +0200
+++ /tmp/femtorv32_petitbateau.v    2023-05-23 06:49:23.115789282 +0200
@@ -279,6 +279,7 @@
       end
    end
 
+   localparam EXECUTE_bit              = 3;
    /***************************************************************************/
    // The FPU 
    /***************************************************************************/
@@ -537,7 +538,6 @@
    localparam FETCH_INSTR_bit          = 0;
    localparam WAIT_INSTR_bit           = 1;
    localparam DECOMPRESS_GETREGS_bit   = 2;   
-   localparam EXECUTE_bit              = 3;
    localparam WAIT_ALU_OR_MEM_bit      = 4;
    localparam WAIT_ALU_OR_MEM_SKIP_bit = 5;

```

JamesTimothyMeech, 2023-06-03:
Thanks for the fix @trabucayre. I appreciate you taking the time to look at my issue!

enjoy-digital, 2023-06-06:
Thanks @trabucayre for looking at this and fixing/improving petitbateau! Let's wait that @BrunoLevy merges your PR and we'll be able to close this one here.

enjoy-digital, 2023-06-06:
Ah OK, thanks @trabucayre for the additional info.

enjoy-digital, 2026-10-21:
Closing since this should now be fixed and there was no further feedback. Feel free to reopen if the issue is still seen.

closed 2026-10-21 by enjoy-digital