enjoy-digital/litex #1698: How to define SYNTHESIS w/o toolchain option

xhackerustc, 2023-05-27:
Hi,

I successfully brought up linux-on-litex for Sipeed Tang Nano 20K fpga development board, I want to mainline the support. But I'm blocked by one issue: urandom is not supported, so I hacked the vexriscv_smp to manually add `define SYNTHESIS. Obviously this isn't accepted in mainline. After searching the issues, I found this issue has been seen, and a pull: https://github.com/enjoy-digital/litex/pull/1392 solved it by using toolchain's cmdline option in Quartus. So but this solution isn't available in my case. Is there any other solution for this case?

Thanks in advance

enjoy-digital, 2023-05-30:
Hi @xhackerustc,

thanks for the feedback. I'll try to see with the author of Tang Nano 20 LiteX/Linux support how it has been done for this demo (https://twitter.com/SipeedIO/status/1659173147340423169): Clean fix or workaround.

I'll also do some tests to see how to add the `define SYNTHESIS with Gowin's toolchain.

enjoy-digital, 2023-06-15:
Hi @xhackerustc,

it's not yet clear to me how to do a define with Gowin toolchain (haven't found the info after quickly looking at the doc or other projects). So I did it like this: https://github.com/enjoy-digital/litex/commit/56927ee73ab66216681cdc6fe131bb2fd96065f0 We could improve in the future if we understand how to define SYNTHESIS with Gowin toolchain.

closed 2023-06-15 by enjoy-digital