enjoy-digital/litex #1700: Memory block defined by Litex

ztachip, 2023-05-28:
I have some RTL code that I am trying to run under Litex framework so that my design is portable between different FPGA. 
But I need to instantiate some dual port memory block.
Does Litex has a way for me to instantiate these dual port memory block so that I dont have to rely on FPGA vector library which would break the portability.
Thanks

enjoy-digital, 2023-05-30:
Hello @ztachip,

Migen's Memory that is used in LiteX is portable between the different FPGA vendor and can be correctly inferred as dual port BRAM, ex: https://github.com/m-labs/migen/blob/master/examples/basic/memory.py.

You can also search in LiteX project for Memory for more examples. If you don't find a convenient example, I'll try to help you further.

ztachip, 2023-05-30:
Wow. This is so awesome.
Do you have some links explaining the blocks you have available to help people porting FPGA design?
Im also looking for blocks such as PLL,...
Thanks
Vuong

enjoy-digital, 2023-05-30:
You can find the PLL modules here: https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/clock and example of use in the LiteX-Boards targets: https://github.com/litex-hub/litex-boards/tree/master/litex_boards/targets

enjoy-digital, 2023-05-30:
Here is a simple example of code to create a dual port RAM with Migen. You'll then be able to use the read/write ports from your logic.

```python3
from migen import *

class DualPortRAM(Module):
    def __init__(self, width=32, depth=1024):  # defaults to a 32-bit wide, 1024-depth memory block
        self.specials.mem = Memory(width, depth)
        
        # Port 1
        self.specials.wrport1 = self.mem.get_port(write_capable=True, we_granularity=8)
        self.specials.rdport1 = self.mem.get_port(has_re=True)

        # Port 2
        self.specials.wrport2 = self.mem.get_port(write_capable=True, we_granularity=8)
        self.specials.rdport2 = self.mem.get_port(has_re=True)

```

closed 2023-05-30 by ztachip