enjoy-digital/litex #1704: PicolibC is likely not configured correctly for FemtoRV

JamesTimothyMeech, 2023-06-02:
When running `python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant gracilis --variant a7-100 --toolchain vivado --load --build`

Meson is passed these settings:

`meson /home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data \
        -Dmultilib=false \
        -Dpicocrt=false \
        -Datomic-ungetc=false \
        -Dthread-local-storage=false \
        -Dio-long-long=true \
        -Dformat-default=integer \
        -Dincludedir=picolibc/riscv64-unknown-elf/include \
        -Dlibdir=picolibc/riscv64-unknown-elf/lib \
        --cross-file cross.txt
WARNING: Unknown CPU family riscv, please report this at https://github.com/mesonbuild/meson/issues/new`

and sets up picolib like this:

`Configuring picolibc.h using configuration
Build targets in project: 9
NOTICE: Future-deprecated features used:
 * 0.56.0: {'meson.source_root'}
 * 0.58.0: {'meson.get_cross_property'}

picolibc 1.7.9

  User defined options
    Cross files         : cross.txt
    includedir          : picolibc/riscv64-unknown-elf/include
    libdir              : picolibc/riscv64-unknown-elf/lib
    atomic-ungetc       : false
    format-default      : integer
    io-long-long        : true
    multilib            : false
    picocrt             : false
    thread-local-storage: false`

which leads to this variables.mak:

`PACKAGES=libc libcompiler_rt libbase libfatfs liblitespi liblitedram libliteeth liblitesdcard liblitesata bios
PACKAGE_DIRS=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libc /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libcompiler_rt /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libbase /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libfatfs /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitespi /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitedram /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libliteeth /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitesdcard /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitesata /home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/bios
LIBS=libc libcompiler_rt libbase libfatfs liblitespi liblitedram libliteeth liblitesdcard liblitesata
TRIPLE=riscv64-unknown-elf
CPU=femtorv
CPUFAMILY=riscv
CPUFLAGS=-march=rv32i2p0_mc   -mabi=ilp32 -D__femtorv__
CPUENDIANNESS=little
CLANG=0
CPU_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/cores/cpu/femtorv
SOC_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc
PICOLIBC_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/pythondata-software-picolibc/pythondata_software_picolibc/data
COMPILER_RT_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/pythondata-software-compiler_rt/pythondata_software_compiler_rt/data
export BUILDINC_DIRECTORY
BUILDINC_DIRECTORY=/home/meechy/Casino/Mine/build/digilent_arty/software/include
LIBC_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libc
LIBCOMPILER_RT_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libcompiler_rt
LIBBASE_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libbase
LIBFATFS_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libfatfs
LIBLITESPI_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitespi
LIBLITEDRAM_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitedram
LIBLITEETH_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/libliteeth
LIBLITESDCARD_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitesdcard
LIBLITESATA_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/liblitesata
BIOS_DIRECTORY=/home/meechy/Casino/Enjoy_Digital/litex/litex/soc/software/bios
LTO=0
BIOS_CONSOLE_FULL=1`

which seems wrong because it specifies a 64 bit compiler but 32 bit arch and abi flags. Could this meson setup issue explain the `undefined reference to __extendsfdf2'` errors I have been getting when trying to compile code for this platform?

enjoy-digital, 2023-06-06:
Hello @JamesTimothyMeech,

this seems similar to https://github.com/enjoy-digital/litex/issues/1624, can you do a test with the toolchain recommended by AEW2015 there?

Are you also using Ubuntu 22.04?

I'll also try to do some tests myself on a Ubuntu 22.04 machine very soon.

JamesTimothyMeech, 2023-06-06:
I'm using Ubuntu 20.04 on WSL. I'll try the other toolchain an post my results here