enjoy-digital/litex #1705: Three states spi-sdcard no longer works (assertion fail)

BrunoLevy, 2023-06-03:
`SOC.add_spi_sdcard(tristate=True)` fails  

```
  File "/home/blevy/Programming/learn-fpga/LiteX/boards/radiona_ulx3s_ex.py", line 121, in main
    soc.add_spi_sdcard(with_tristate=True)
  File "/home/blevy/Programming/LiteX/litex/litex/soc/integration/soc.py", line 1816, in add_spi_sdcard
    self.add_module(name=f"{name}_tristate", module=tristate)
  File "/home/blevy/Programming/LiteX/litex/litex/gen/fhdl/module.py", line 54, in add_module
    assert isinstance(module, Module)
AssertionError
```

The problem seems to come from the following line in `litex/litex/soc/integration/soc.py`
```
self.add_module(name=f"{name}_tristate", module=tristate)
```

enjoy-digital, 2023-06-04:
Hi @BrunoLevy,

this should be fixed with https://github.com/enjoy-digital/litex/commit/a1106b997e33c2783a9088bfbc87e34b8b0de54c.

Florent

enjoy-digital, 2023-06-06:
Closing since should be fixed, please re-open @BrunoLevy if not.

closed 2023-06-06 by enjoy-digital