enjoy-digital/litex #1708: Signal always  being reset ?

OJthe123, 2023-06-09:
Hi.
I just started with migen/LiteX and FPGA.

I want to add an UART to colorlite.py
So I copied and modified the parts from ios_stream.py. Sending four bytes over UART works fine. But it keep on sending as far as clock allows it.
So I tried to add a counter to slow it down. But I cannot get it to work.
Maybe this goal is too high for a beginner? Maybe you can give me a little help :-)

This is the part I added into colorlite.py

class IOStreamer(Module):
    """Stream bytes from memory over UART"""
    def __init__(self, pad, sys_clk_freq, baudrate):
        from litex.soc.cores.uart import RS232PHYTX

        # Timer/Counter
        self.counter = Signal(max=sys_clk_freq +1)
        
        # UART
        pads = Record([("tx", 1)])
        self.comb += pad.eq(pads.tx)
        phy = RS232PHYTX(pads, int((baudrate/sys_clk_freq)*2**32))
        self.submodules += phy

        # Memory
        mem  = Memory(8, 4, init=[0xAA, 0xBB, 0xCC, 0xDD])
        port = mem.get_port()
        
        
        self.specials += mem, port
        self.comb += phy.sink.valid.eq(1)
        self.comb += If(self.counter == int(12e6), self.counter.eq(0), phy.sink.data.eq(port.dat_r) ) 
        #self.comb += phy.sink.data.eq(port.dat_r)
        
        self.sync += self.counter.eq(self.counter+1)
        self.sync += If(phy.sink.ready, port.adr.eq(port.adr + 1))
        

and in the SoCMini section

# Servos -----------------------------------------------------------------------------------
        from litex.soc.cores.pwm import PWM
        for n in range(4):
            servo_pad = platform.request("servo", n)
            servo     = PWM(servo_pad)
            setattr(self.submodules, f"servo{n}", servo)

        # IO_Streamer ------------------------------------------------------------------------------
        self.submodules.iostreamer = IOStreamer(platform.request("gpio", 2), sys_clk_freq, baudrate=9600)

without my counter it is sending AA BB CC DD over UART. with my counter just zeros.
Where is my fault?

A look into the verilog code gives : 

```
...
reg    [25:0] iostreamer_counter = 26'd0;
...
always @(*) begin
    iostreamer_sink_payload_data <= 8'd0;
    iostreamer_counter <= 26'd0;
    if ((iostreamer_counter == 24'd12000000)) begin
        iostreamer_counter <= 1'd0;
        iostreamer_sink_payload_data <= iostreamer_dat_r;
    end
end
...
```

wouldn't that mean the counter will always be reset to zero and never reach 12000000 ?

enjoy-digital, 2023-06-09:
Hi @OJthe123,

one error I can see:
`self.comb += If(self.counter == int(12e6), self.counter.eq(0), phy.sink.data.eq(port.dat_r) ) `
-> This should be synchronous:
`self.sync += If(self.counter == int(12e6), self.counter.eq(0), phy.sink.data.eq(port.dat_r) ) `

OJthe123, 2023-06-11:
```
class IOStreamer(Module):
    """Stream bytes from memory over UART"""
    def __init__(self, pad, sys_clk_freq, baudrate):
        from litex.soc.cores.uart import RS232PHYTX

        # Timer/Counter
        counter = Signal(max=sys_clk_freq +1)
        
        # UART
        pads = Record([("tx", 1)])
        self.comb += pad.eq(pads.tx)
        phy = RS232PHYTX(pads, int((baudrate/sys_clk_freq)*2**32))
        self.submodules += phy

        # Memory
        mem  = Memory(8, 4, init=[0xAA, 0xBB, 0xCC, 0xDD])
        port = mem.get_port()
        
        
        self.specials += mem, port
        
        self.sync += counter.eq(counter+1)
        #self.sync += phy.sink.valid.eq(0)
        self.sync += If(phy.sink.ready, phy.sink.valid.eq(0), phy.sink.ready.eq(0))
        self.sync += If(counter == int(49e6) , counter.eq(0), phy.sink.data.eq(port.dat_r), port.adr.eq(port.adr + 1), phy.sink.valid.eq(1) )

```

This, I thought should work, but it doesn't :smile: 
In between I played arround but it is sending garbage...the best try was sending "AA AA BB CC DD AA AA AA AA BB ..... and so on.
At least the one swcond delay worked. Another sequence was 20 bytes of AA, one second break, 20 bytes of AA :smile: 
I am getting frustrated...but it's okay for me :wink: 

RS232PHYTX waits for sink.valid to become 1. then it takes the data and sends it through the FSM. Last action in the FSM is to set sink.ready.eq(1).
So what is wrong here? 
I have the feeling there is more going on in the background than I know, yet. 
But I will learn hopefully :smile: