enjoy-digital/litex #1709: More than one DFU capable USB device found! Error

JamesTimothyMeech, 2023-06-09:
When running: `python3 -m litex_boards.targets.gsd_orangecrab --cpu-type femtorv --cpu-variant petitbateau --load --device 85F --sdram-device MT41K64M16 --ecppack-compress --with-spi-sdcard
`
I get this error:
`dfu-suffix (dfu-util) 0.11

Copyright 2011-2012 Stefan Schmidt, 2013-2020 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Suffix successfully added to file
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1209
Match product ID from file: 5af0
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device`

enjoy-digital, 2023-06-09:
Hi @JamesTimothyMeech,

as suggested by the error report, you probably have another DFU device connected on your computer. You'll probably have to disconnect it or specify the OrangeCrab DFU ID when loading (so modify the code for this since current code assume the board is the only DFU connected).

JamesTimothyMeech, 2023-06-09:
My comment on my PR shows that the orangecrab is the only device connected:

enjoy-digital, 2023-06-09:
Ok thanks, I'll have to do some tests since the OrangeCrab is not the only DFU device supported, so we can't force it in LiteX (or would eventually have to do it in LiteX-Boards in the OrangeCrab target).

JamesTimothyMeech, 2023-06-09:
That makes sense, I only realised that my fix was a hack after I had made the PR!

JamesTimothyMeech, 2023-06-09:
If it works for most people my error could be because I am using a very old version of dfu-utils

enjoy-digital, 2023-06-09:
Can you try to use a newer version of dfu-utils to see if that's the reason?

JamesTimothyMeech, 2023-06-09:
If I figure out how to install a new version of dfu-utils on my Mac I’ll post the results on here. I’ve been using version 0.11 because that is the version that both MacPorts and Brew install

JamesTimothyMeech, 2023-06-09:
Any idea where I can get a newer version for Mac? I installed the one from here but it is still 0.11: https://dfu-util.sourceforge.net/

JamesTimothyMeech, 2023-06-30:
Same error appears when using the default dfu-util install by Ubuntu 22.04:
```
dfu-suffix (dfu-util) 0.9

Copyright 2011-2012 Stefan Schmidt, 2013-2014 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Suffix successfully added to file
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1209
Match product ID from file: 5af0
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device
```

JamesTimothyMeech, 2023-06-30:
There is a related fix here on the orangecrab repo: https://github.com/orangehttps://github.com/orangecrab-fpga/orangecrab-hardware/issues/48crab-fpga/orangecrab-hardware/issues/48

JamesTimothyMeech, 2023-06-30:
Just adding the "--alt", "0", part works for me if the "--device", "1209:5af0" messes up the download for the other boards

trabucayre, 2023-07-28:
Sorry for delay. Yes as for ulx3s with dfu bootloader, orangeCrab has more than one alt point and you have to use 0 to tell dfu-utils which one to use.
this [line](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/gsd_orangecrab.py#L232) must be changed by:
```
return DFUProg(vid="1209", pid="5af0", alt=0)
```

enjoy-digital, 2023-08-28:
Fixed with https://github.com/litex-hub/litex-boards/commit/5799c352470daeb96541285af6cea0aa6827ef9a, thanks @trabucayre.

closed 2023-08-28 by enjoy-digital