enjoy-digital/litex #1716: Pip issues when installing Litex

GuzTech, 2023-06-22:
I tried to run `./litex_setup.py --init --user -config=full` on a fully updated Arch box and during the installation of Git repositories step, I get the following error:

```bash
./litex_setup.py --install --user --config=full
          __   _ __      _  __         
         / /  (_) /____ | |/_/         
        / /__/ / __/ -_)>  <           
       /____/_/\__/\__/_/|_|         
     Build your hardware, easily!      
          LiteX Setup utility.         

[   0.001] LiteX Setup auto-update...
[   0.403] LiteX Setup is up to date.
[   0.404] Installing Git repositories...
[   0.404] ------------------------------
[   0.404] Installing migen Git repository...
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try 'pacman -S
    python-xyz', where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Arch-packaged Python package,
    create a virtual environment using 'python -m venv path/to/venv'.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-Arch packaged Python application,
    it may be easiest to use 'pipx install xyz', which will manage a
    virtual environment for you. Make sure you have python-pipx
    installed via pacman.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Traceback (most recent call last):
  File "/home/oguz/Software/litex/./litex_setup.py", line 480, in <module>
    main()
  File "/home/oguz/Software/litex/./litex_setup.py", line 460, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
  File "/home/oguz/Software/litex/./litex_setup.py", line 285, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pip install --editable . {options}".format(
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/usr/bin/python3" -m pip install --editable . --user' returned non-zero exit status 1.
```

So apparently some things with `pip` have changed and the recommended way to install packages is to use `pipx` instead. I have modified the `litex_setup.py` script to use `pipx` but then I get the following issue:

```bash
/litex_setup.py --install --user --config=full
          __   _ __      _  __         
         / /  (_) /____ | |/_/         
        / /__/ / __/ -_)>  <           
       /____/_/\__/\__/_/|_|         
     Build your hardware, easily!      
          LiteX Setup utility.         

[   0.001] Installing Git repositories...
[   0.001] ------------------------------
[   0.001] Installing migen Git repository...

No apps associated with package migen or its dependencies. If you are attempting to install a library, pipx should not be used. Consider using pip or a similar tool instead.
Traceback (most recent call last):
  File "/home/oguz/Software/litex/./litex_setup.py", line 484, in <module>
    main()
  File "/home/oguz/Software/litex/./litex_setup.py", line 464, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
  File "/home/oguz/Software/litex/./litex_setup.py", line 289, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pipx install --editable . {options}".format(
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/usr/bin/python3" -m pipx install --editable . ' returned non-zero exit status 1.
```

I can actually install packages with `pipx`, but since (in this case) Migen is not an application, `pipx` returns with error code 1. So I have modified the script further to not check the return code and with that change it now also installs properly. The relevant hacky changes to `litex_setup.py` can be found here: https://github.com/GuzTech/litex/compare/master...pipx-patch

trabucayre, 2023-06-22:
This behavior is more or less common to all distro since few month.
One solution is to add `--break-system-packages` after `pip install`
Another solution is to create a file **$HOME/.config/pip/pip.conf** with

```
[global]
break-system-packages = true
```

Second solution is great because is not limited to `litex_setup.py` but all users have to create this file, so maybe a small update to add argument looks better here.

GuzTech, 2023-06-22:
How wise is it to potentially break system packages though? I do not know how realistic the odds are that some users will have issues if they use the `break-system-packages` method.

trabucayre, 2023-06-22:
The main risk is to have for the same package one installed by the pkg manager and one by pip, with different version. Depending directory order with PYTHON_PATH or something similar, users may have issue due to this situation.
I use a *gentoo* distro (where everything is built from sources), I have already observed issues due to a pkg installed by pip.

sensille, 2023-06-26:
Same here, but when trying to fix litex after a system update. Instead of calling litex_setup installing meson via pacman fixed my install.

curiousmuch, 2023-07-08:
Does anyone have this working in a virtual environment? I tried earlier, but I couldn't get it work.

AndrewD, 2023-08-06:
I found with arch linux that --user is no longer allowed, but --editable (development) installs are actually still installed per user so not specifying  --user fixed this for me.

josuah, 2023-09-04:
This sounds like the effect of https://peps.python.org/pep-0668/ which various distributions start slowly rolling out.

This pushes virtual environments down the throat of the user, but solves other problems for the user as well.

Not sure this can be helped at the level of LiteX.

Running `python -m venv ~/litex/.venv` then `. ~/litex/.venv/bin/activate` before `./litex_setup.py` might be the fix.

If wanting to forget the fact that you use virtual environments, one way is to run `python -m venv ~/.venv` and add this to the `~/.profile`:

```shell
if [ -f "$HOME/.venv/bin/activate" ]; then
    . "$HOME/.venv/bin/activate"
fi
```

Then every `pip` command would install things in ~/.venv instead of ~/.local, but besides than that, your system might work as it was before https://peps.python.org/pep-0668/ got enforced.

nickoe, 2025-04-09:
Hi @josuah 

How do we make this work? I just tried this today in a new directory. 

Tried to do:
```
python -m venv --prompt litex_2025 .venv/
...
. .venv/bin/activate
... prompt now shows
(litex_2025) [litex_2025_sipeed]$
... doing
./litex_setup.py --init --install
... fails with
[  17.018] Installing litex Git repository...
Obtaining file:///home/nickoe/litex

[notice] A new release of pip is available: 24.3.1 -> 25.0.1
[notice] To update, run: pip install --upgrade pip
ERROR: file:///home/nickoe/litex does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Traceback (most recent call last):
  File "/home/nickoe/litex_2025_sipeed/./litex_setup.py", line 497, in <module>
    main()
    ~~~~^^
  File "/home/nickoe/litex_2025_sipeed/./litex_setup.py", line 477, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nickoe/litex_2025_sipeed/./litex_setup.py", line 290, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pip install {editable} . {options}".format(
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        python3  = sys.executable,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
        editable = "--editable" if repo.editable else "",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        options  = "--user"     if user_mode else "",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        ), shell=True)
        ^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/subprocess.py", line 421, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/home/nickoe/litex_2025_sipeed/.venv/bin/python3" -m pip install --editable . ' returned non-zero exit status 1.
```

I have no idea why it refers to ~/litex

trabucayre, 2025-04-09:
@nickoe True: the venv contains a .gitignore file, and consequently clone are performed in parent directory.
So the solution, in your venv directory, is to create a subdirectory and uses `litex_setup.py` here:
```bash
python -m venv --prompt litex_2025 .venv/
. .venv/bin/activate
mkdir litex
cd litex
wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
./litex_setup.py --init --install --user (--user to install to user directory) --config=(minimal, standard, full)
```