enjoy-digital/litex #1719: no uart when changing sys-clock-freq

gustavosousa2208, 2023-06-26:
im using sipeed tang nano 9k

when i run `python3 -m litex_boards.targets.sipeed_tang_nano_9k --sys-clk-freq=108000000 --build --load`
it runs ok, but no uart, only with 27e6 clock

`litex_term /dev/ttyUSB1 --speed 115200 --kernel=./demo.bin` returns nothing

is there a way to change the serial divider/baud rate?

trabucayre, 2023-06-26:
Not sure issue is related to the baudrate (and by using `--sys-clk-freq=108000000` the baudrate divider is automatically adapted to this frequency).
I have build litex for this board and checked timing report (*build/sipeed_tang_nano_9k/gateware/impl/pnr/project_tr_content.html*): **FMAX** is around 45MHz. This may explain why you have nothing.
I'm not sure flash and hyperram are able to works at 108MHz too.
`ledChaser` may indicate gateware is build and loaded but it's not an information about softcore's healty.

enjoy-digital, 2023-06-26:
Hi @gustavosousa2208,

as pointed by @trabucayre, the `sys-clk-freq `provided is too high for this board. LiteX is persmissive on this aspect and lets the user configure `sys-clk-freq` but we then expect user to check timing reports to see if timings are met.

closed 2023-06-27 by gustavosousa2208