enjoy-digital/litex #1722: .repl and .resc files for Renode cannot be created from Litex

fdsgdshn, 2023-07-03:
Hi, 
I'm encountering an error while trying to generate the .resc and .repl files expected to be created by Litex for Renode, using Litex's Python files. How can I overcome this issue? Could you provide some guidance? 

Below, I have attached detailed information regarding the error I'm experiencing.
Thank you!

I'm getting an error indicating that the csr.json file does not contain the "config_cpu_count" parameter. The error occurs in litex/build/lib/litex/tools/litex_json2renode.py, specifically in line 217, where it tries to access csr['constants']['config_cpu_count'], and in line 1080 resulting in a KeyError: 'config_cpu_count'.

nickoe, 2023-08-21:
How exactly are you running this?

enjoy-digital, 2024-01-12:
Closing since no feedback. @fdsgdshn Please provide more info if you still have questions.

closed 2024-01-12 by enjoy-digital