enjoy-digital/litex #1723: Problem when adding my VHDL module to litex

pratipm, 2023-07-03:
This may be a migen issue, but I am having problem when trying to add my VHDL module to litex SOC.
In the class  BaseSOC(SocCore), I am trying to add my module, but when adding a Signal like this:

adc_clock = Signal
self.comb += adc_clock.eq(0)

Litex build giving this error:
Traceback (most recent call last):
  File "/home/pratipm/litex/litex-boards/litex_boards/targets/atlys_dso.py", line 276, in <module>
    main()
  File "/home/pratipm/litex/litex-boards/litex_boards/targets/atlys_dso.py", line 261, in main
    soc = BaseSoC(
  File "/home/pratipm/litex/litex-boards/litex_boards/targets/atlys_dso.py", line 231, in __init__
    self.comb += adc_clock.eq(0)
TypeError: _Value.eq() missing 1 required positional argument: 'r'

What is this 'required positional argument 'r'?  Cannot find anything in the Migen documents.

Thanks.

trabucayre, 2023-07-03:
The problem is with:
```
adc_clock = Signal
```
instead of:
```
adc_clock = Signal()
```
without parenthesis it's not an instance of class `Signal`:

```
>>> from migen import *
>>> t1 = Signal
>>> type(t1)
<class 'type'>
>>> t2=Signal()
>>> type(t2)
<class 'migen.fhdl.structure.Signal'>
```

pratipm, 2023-07-04:
OOPS! Did not notice that and I have been looking at the code for hours!
Thanks you so much.

closed 2023-07-04 by pratipm