enjoy-digital/litex #1725: ttyUSBX missing when trying to use litex_term to load code

DanaelCarbonneau, 2023-07-06:
I'm currently working with an OrangeCrab FPGA, and trying to load code after the boot, following the indications given on `litex/litex/soc/software/demo/README.md` (build and load over LiteX-Term).

The command that is suggested (`litex_term /dev/ttyUSBX --kernel=demo.bin`) doesn't work, because there is no ttyUSBX, and the only port that there seems to be (according to dmesg) is the one already used by the console (ttyACM0). I assume that I'm not building my soc correctly, how do I fix that ?

The command I use to build the SOC : 
`python3 -m litex_boards.targets.gsd_orangecrab --uart-name=usb_acm --build --load`

trabucayre, 2023-07-07:
`/dev/ttyUSB` is the base name for all USB<->serial adapters,  This base name is appended by a number starting at 0, incremented each time a new adapter is plugged and decremented when unplugged. **X** is generic to said "replace it by the number of the interface to use as serial adapter".
Indeed, this number depends on the board and each user situation. For example when you use an *arty* board you have 2 `ttyUSB` devices (0: jtag, 1: serial), but if, before plugging you arty, an adapter is alrready plugged your *arty* will have `ttyUSB1` and `ttyUSB2`.
For you specific situation, there is no USB<->serial but a cdc acm device, the Linux kernel, instead of using `ttyUSB` uses `ttyACM` name.
So you have to replace `/dev/ttyUSBX` by `/dev/ttyACM0`.

DanaelCarbonneau, 2023-07-07:
Great ! It worked ! Thanks !

closed 2023-07-07 by DanaelCarbonneau

gsomlo, 2023-07-07:
On Thu, Jul 06, 2023 at 10:22:03PM -0700, Gwenhael Goavec-Merou wrote:
> /dev/ttyUSB is the base name for all USB<->serial adapters, This base name is

FWIW, I run `dmesg -w` in a separate terminal window, and see what
device name(s) show up when I plug in any of the various fpga dev
boards I have, which then tells me what tty device(s) to try for the
litex console :)

trabucayre, 2023-07-07:
I use most of the time the same technic.