enjoy-digital/litex #1726: --sys-clk-freq=48e6 does not work on Arty

JamesTimothyMeech, 2023-07-06:
When I run ```python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant gracilis --variant a7-100 --toolchain vivado --with-spi-sdcard --sdcard-adapter digilent --build --load --sys-clk-freq=48e6
```
the functionality to set the clock frequency does not work and throws and error:

```
james@james-Laptop-13th-Gen-Intel-Core:~/Desktop/Casino/FPGA-System-on-Chip-Firmware/LiteX$ python3 -m litex_boards.targets.digilent_arty --cpu-type femtorv --cpu-variant gracilis --variant a7-100 --toolchain vivado --with-spi-sdcard --sdcard-adapter digilent --build --load --sys-clk-freq=48e6
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 48.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 eth of 25.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x of 192.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut3 sys4x_dqs of 192.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut4 idelay of 200.00MHz (+-10000.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2023-07-06 15:34:44)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a100tcsg324-1.
INFO:SoC:System clock: 48.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU femtorv added.
INFO:SoC:CPU femtorv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU femtorv setting reset address to 0x00000000.
INFO:SoC:CPU femtorv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x01000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x01000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:CSR Bridge csr added.
INFO:SoCBusHandler:csr Region added at Origin: 0x82000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:csr added as CSR Master.
INFO:SoCBusHandler:Interconnect: InterconnectShared (1 <-> 4).
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCCSRHandler:ddrphy CSR allocated at Location 1.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:leds CSR allocated at Location 3.
INFO:SoCCSRHandler:sdram CSR allocated at Location 4.
INFO:SoCCSRHandler:spisdcard CSR allocated at Location 5.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 6.
INFO:SoCCSRHandler:uart CSR allocated at Location 7.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Finalized SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False
Bus Regions: (4)
rom                 : Origin: 0x00000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False
sram                : Origin: 0x01000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False
main_ram            : Origin: 0x40000000, Size: 0x10000000, Mode: RWX, Cached: True Linker: False
csr                 : Origin: 0x82000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
Bus Masters: (1)
- cpu_bus0
Bus Slaves: (4)
- rom
- sram
- main_ram
- csr
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
CSR Locations: (8)
- ctrl           : 0
- ddrphy         : 1
- identifier_mem : 2
- leds           : 3
- sdram          : 4
- spisdcard      : 5
- timer0         : 6
- uart           : 7
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 221, in <module>
    main()
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex-boards/litex_boards/targets/digilent_arty.py", line 210, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex/litex/soc/integration/builder.py", line 332, in build
    self.soc.finalize()
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex/litex/soc/integration/soc.py", line 1301, in finalize
    Module.finalize(self)
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/james/Desktop/Casino/Enjoy_Digital/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex/litex/soc/cores/clock/xilinx_s7.py", line 31, in do_finalize
    config = self.compute_config()
  File "/home/james/Desktop/Casino/Enjoy_Digital/litex/litex/soc/cores/clock/xilinx_common.py", line 109, in compute_config
    raise ValueError("No PLL config found")
ValueError: No PLL config found

```

enjoy-digital, 2023-07-06:
Hi @JamesTimothyMeech,

that's a limitation of the PLL primitive. If 48MHz is mandatory, adding a specific PLL to generate it should work.

JamesTimothyMeech, 2023-07-06:
I see, would I do that in the ```litex_boards.targets.digilent_arty``` file or do I need to put the PLL in the .v file which LiteX pulls in for the SoC?

enjoy-digital, 2023-07-06:
You can do that directly in the target file, just add another `S7PLL`.

JamesTimothyMeech, 2023-07-06:
Thanks, I will close this now as I fixed my problem another way https://github.com/BrunoLevy/learn-fpga/discussions/110