enjoy-digital/litex #1732: booting hang

jwfaye, 2023-07-19:
Hello,
I encoutered a problem that I never seen before. I build a CPU with no main ram by default at 0x4000_0000 in litex. I got a sram and I want to load code from its starting address. When I do the loading it hangs after saying that the booting process. When I integrate the main-ram at 0x4000_0000 It boot and execute the program as usual. I tried to instantiate the starting address of my integrated sram but the problem still the same. 

 reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jul 19 2023 18:03:54
 BIOS CRC passed (0e36fb36)

 LiteX git sha1: 69c6fa11

--=============== SoC ==================--
CPU:            FemtoRV-STANDARD @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32.0KiB
SRAM:           16.0KiB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x01000000 (11064 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.

jwfaye, 2023-07-20:
Hello.
I compile it for the sram location. I checked the .map file to see where the _start() entry point is located and it corresponds to the starting address of the sram. Is there any other detail that I missed ?

tcal-x, 2023-07-20:
I just remembered another gotcha.   The BIOS code lives in ROM, but it needs to use SRAM for its storage.  When you upload your app at the base of SRAM, you might be corrupting the BIOS storage and cause a hang before control is handed off to demo.bin.   Perhaps try splitting the SRAM into two regions.   16KiB isn't much, so I'm not sure if there's space for both BIOS working memory and your ~11KiB app simultaneously.

jwfaye, 2023-07-25:
Hello,
Thank you. You are right about the origin of the problem. 
I had a look at the linker file of the bios : 
`INCLUDE generated/output_format.ld
ENTRY(_start)

INCLUDE generated/regions.ld

SECTIONS
{
	.text :
	{
		_ftext = .;
		/* ARM (Quicklogic EOS-S3, Gowin EMCU) interrupt vector tables use section name "isr_vector" */
		/* - such a table has to be placed at start */
		KEEP(*(.isr_vector))
		/* Make sure crt0 files come first, and they, and the isr */
		/* don't get disposed of by greedy optimisation */
		*crt0*(.text)
		KEEP(*crt0*(.text))
		KEEP(*(.text.isr))

		*(.text .stub .text.* .gnu.linkonce.t.*)
		_etext = .;
	} > rom

	.rodata :
	{
		. = ALIGN(8);
		_frodata = .;
		*(.rodata .rodata.* .gnu.linkonce.r.*)
		*(.rodata1)
		*(.got .got.*)
		*(.toc .toc.*)

		/* Make sure the file is aligned on disk as well
		   as in memory; CRC calculation requires that. */
		FILL(0);
		. = ALIGN(8);
		_erodata = .;
	} > rom

	.commands :
	{
		PROVIDE_HIDDEN (__bios_cmd_start = .);
		KEEP(*(.bios_cmd))
		PROVIDE_HIDDEN (__bios_cmd_end = .);
	} > rom

	.init :
	{
		PROVIDE_HIDDEN (__bios_init_start = .);
		KEEP(*(.bios_init))
		PROVIDE_HIDDEN (__bios_init_end = .);
	} > rom

	.data :
	{
		. = ALIGN(8);
		_fdata = .;
		*(.data .data.* .gnu.linkonce.d.*)
		*(.data1)
		*(.sdata .sdata.* .gnu.linkonce.s.*)

		/* Make sure the file is aligned on disk as well
		   as in memory; CRC calculation requires that. */
		FILL(0);
		. = ALIGN(8);
		_edata = .;
	} > sram AT > rom

	.bss :
	{
		. = ALIGN(8);
		_fbss = .;
		*(.dynsbss)
		*(.sbss .sbss.* .gnu.linkonce.sb.*)
		*(.scommon)
		*(.dynbss)
		*(.bss .bss.* .gnu.linkonce.b.*)
		*(COMMON)
		. = ALIGN(8);
		_ebss = .;
		_end = .;
	} > sram

	/DISCARD/ :
	{
		*(.eh_frame)
		*(.comment)
	}
}

PROVIDE(_fstack = ORIGIN(sram) + LENGTH(sram));

PROVIDE(_fdata_rom = LOADADDR(.data));
PROVIDE(_edata_rom = LOADADDR(.data) + SIZEOF(.data));`
Here is there an obligation to have the .bss and .data sections at the same address location for my code ? 
I mean, I create a memory region on the SRAM, I first loaded the .text and .rodata section at SRAM_start+offset and .bss and .data at SRAM_start. Then I was wondering if the .data .bss sections were correlated and then I tried to put those section at the following of the .text and .rodata sections this results at corruption of the bootloader.

jwfaye, 2023-07-25:
Bios : 
(litex) ~/femtorv_soc/software/bios$ readelf -t bios.elf
Il y a 19 en-têtes de section, débutant à l'adresse de décalage 0x4541c:

En-têtes de section :
  [Nr] Nom
       Type            Adr      Décala.Taille ES   LN Inf Al
       Fanions
  [ 0] 
       NULL            00000000 000000 000000 00   0   0  0
       [00000000]: 
  [ 1] .text
       PROGBITS        00000000 000074 0045ec 00   0   0  4
       [00000007]: WRITE, ALLOC, EXEC
  [ 2] .rodata
       PROGBITS        000045ec 004660 001724 00   0   0  4
       [00000002]: ALLOC
  [ 3] .srodata
       PROGBITS        00005d10 005d84 00000c 00   0   0  4
       [00000002]: ALLOC
  [ 4] .commands
       PROGBITS        00005d1c 005d90 000038 00   0   0  4
       [00000003]: WRITE, ALLOC
  [ 5] .data
       PROGBITS        01000000 005dc8 0000f0 00   0   0  4
       [00000003]: WRITE, ALLOC
  [ 6] .bss
       NOBITS          010000f0 005eb8 000560 00   0   0  4
       [00000003]: WRITE, ALLOC
  [ 7] .riscv.attributes
       RISCV_ATTRIBUTE 00000000 005eb8 000020 00   0   0  1
       [00000000]: 
  [ 8] .debug_line
       PROGBITS        00000000 005ed8 013df5 00   0   0  1
       [00000000]: 
  [ 9] .debug_info
       PROGBITS        00000000 019ccd 00833f 00   0   0  1
       [00000000]: 
  [10] .debug_abbrev
       PROGBITS        00000000 02200c 0031ba 00   0   0  1
       [00000000]: 
  [11] .debug_aranges
       PROGBITS        00000000 0251c8 0005b8 00   0   0  8
       [00000000]: 
  [12] .debug_str
       PROGBITS        00000000 025780 010cfd 01   0   0  1
       [00000030]: MERGE, STRINGS
  [13] .debug_loc
       PROGBITS        00000000 03647d 0060bf 00   0   0  1
       [00000000]: 
  [14] .debug_ranges
       PROGBITS        00000000 03c53c 000b80 00   0   0  1
       [00000000]: 
  [15] .debug_macro
       PROGBITS        00000000 03d0bc 006e2b 00   0   0  1
       [00000000]: 
  [16] .symtab
       SYMTAB          00000000 043ee8 000d10 10  17  97  4
       [00000000]: 
  [17] .strtab
       STRTAB          00000000 044bf8 000764 00   0   0  1
       [00000000]: 
  [18] .shstrtab
       STRTAB          00000000 04535c 0000bf 00   0   0  1
       [00000000]: 

Code : 
(litex) ~/firmware$ readelf -t demo.elf
Il y a 21 en-têtes de section, débutant à l'adresse de décalage 0x2b520:

En-têtes de section :
  [Nr] Nom
       Type            Adr      Décala.Taille ES   LN Inf Al
       Fanions
  [ 0] 
       NULL            00000000 000000 000000 00   0   0  0
       [00000000]: 
  [ 1] .text
       PROGBITS        01001000 000084 001cac 00   0   0  4
       [00000007]: WRITE, ALLOC, EXEC
  [ 2] .rodata
       PROGBITS        01002cac 001d30 000174 00   0   0  4
       [00000002]: ALLOC
  [ 3] .eh_frame
       PROGBITS        01002e20 001ea4 0005f4 00   0   0  4
       [00000002]: ALLOC
  [ 4] .srodata
       PROGBITS        01003414 002498 00000c 00   0   0  4
       [00000002]: ALLOC
  [ 5] .srodata.cst4
       PROGBITS        01003420 0024a4 000004 04   0   0  4
       [00000012]: ALLOC, MERGE
  [ 6] .data
       PROGBITS        01000000 000074 000010 00   0   0  4
       [00000003]: WRITE, ALLOC
  [ 7] .bss
       NOBITS          01000010 000084 000000 00   0   0  1
       [00000003]: WRITE, ALLOC
  [ 8] .debug_info
       PROGBITS        00000000 0024a8 0041ba 00   0   0  1
       [00000000]: 
  [ 9] .debug_abbrev
       PROGBITS        00000000 006662 001ad9 00   0   0  1
       [00000000]: 
  [10] .debug_loc
       PROGBITS        00000000 00813b 00333e 00   0   0  1
       [00000000]: 
  [11] .debug_aranges
       PROGBITS        00000000 00b480 000368 00   0   0  8
       [00000000]: 
  [12] .debug_macro
       PROGBITS        00000000 00b7e8 00641e 00   0   0  1
       [00000000]: 
  [13] .debug_line
       PROGBITS        00000000 011c06 00ac92 00   0   0  1
       [00000000]: 
  [14] .debug_str
       PROGBITS        00000000 01c898 00dc3a 01   0   0  1
       [00000030]: MERGE, STRINGS
  [15] .comment
       PROGBITS        00000000 02a4d2 00002a 01   0   0  1
       [00000030]: MERGE, STRINGS
  [16] .riscv.attributes
       RISCV_ATTRIBUTE 00000000 02a4fc 000020 00   0   0  1
       [00000000]: 
  [17] .debug_ranges
       PROGBITS        00000000 02a51c 000478 00   0   0  1
       [00000000]: 
  [18] .symtab
       SYMTAB          00000000 02a994 000720 10  19  51  4
       [00000000]: 
  [19] .strtab
       STRTAB          00000000 02b0b4 000393 00   0   0  1
       [00000000]: 
  [20] .shstrtab
       STRTAB          00000000 02b447 0000d6 00   0   0  1
       [00000000]: 

Here The .data section of the code starts at the same adress than the .data section of the bios. But it works fine.

jwfaye, 2023-07-28:
The issue is closed as the solution have been provided

closed 2023-07-28 by jwfaye