enjoy-digital/litex #1739: Wishbone L2 cache specs

Nicolas-Gaudin, 2023-08-01:
Hi, 

What are the specifications of the Wishbone L2 cache ? 

I would like to know the type of cache (direct-mapped or set associative).
If set associative, is it possible to change the number of ways and the replacement policy ?

Thanks

josuah, 2023-08-10:
Only a clue rather than a full answer, but you might to want where, and in which configuration the `wishbone.Cache()` class is used:

https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L542

Nicolas-Gaudin, 2023-08-10:
Thanks for the clue, I'd had a look at this class.
From there, we can determine very little about the cache configuration (it's a write-back, and I assume it's a direct cache). 

I am wondering how to modify the cache in order to implement my own L2 cache (generated automatically by the class or not) but with a different configuration.