enjoy-digital/litex #1750: Various Issues with Sipeed Tang Primer 20k

bacintom, 2023-08-20:
Hi,

I'm getting started on FPGAs (apologies in advance for noob behavior).

I've bought a Sipeed Tang Primer 20k ( https://wiki.sipeed.com/hardware/en/tang/tang-primer-20k/primer-20k.html ) and I'm trying to use LiteX on it. The board is already available in litex-boards.

I have a few issues, also found some workarounds (no 'nice' solution though, still I hope it helps), I'll report about them here.
I'm open to any suggestions if I could help improve support for this board.

Sorry it is quite long, I'll try to split into separate parts. Summary :

1. The reported amount of RAM is 256MB, but it should be 128?
2. Memory initialization fails (workaround: change latency settings)
3. SD card init fails (workaround: use spisdcard)
4. Building with --with-video-terminal fails (workaround: disable frequency check).
5. Trying to boot linux-on-litex-vexriscv: so far, stuck at "liftoff!"

------
# Environment

* OS: ubuntu 23.04
* python 3.11.4
* litex updated today (Aug 20 2023) using ./litex_setup.py --update
* toolchain: Gowin_V1.9.8.11_Education_linux
------

# 1. The reported amount of RAM is 256MB, but it should be 128?

Building litex for my board, with the most default settings:

`
python3 -m litex_boards.targets.sipeed_tang_primer_20k --build --load
`

I get the following output from the serial:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Aug 20 2023 13:33:28
 BIOS CRC passed (45f9f79d)

 LiteX git sha1: 1520d0f3

--=============== SoC ==================--
CPU:            VexRiscv @ 48MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
SDRAM:          256.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:       256.0MiB
```

According to the specs ( https://wiki.sipeed.com/hardware/en/tang/tang-primer-20k/primer-20k.html ), my board is supposed to have 1Gbit / 128MB of DDR3. Why does it say 256MB...?

# 2. Memory initialization fails

This is the second part of the serial output:

```
--========== Initialization ============--
Initializing SDRAM @0x40000000...
[... 203 lines cut ...]
[########################################]
Copying rootfs.cpio to 0x41000000 (3781632 bytes)...
[########################################]
Copying opensbi.bin to 0x40f00000 (53640 bytes)...
[########################################]
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

```

Then it is stuck forever at this point.
I also tried without video-terminal. But video-terminal is useful to confirm I didn't just loose the serial connection. Anyway, always the same result.

---
Note: SDCard (not spi) gives:

```
litex> sdcardboot
Booting from SDCard in SD-Mode...
Booting from boot.json...
Booting from boot.bin...
SDCard boot failed.
```
... as expected because sdcard_init fails.

trabucayre, 2023-08-25:
Hi @bacintom 
We have tried your fix for DDR3 with two tang Primer 20k variants:
- one with *SKHynix* chip: **KO**
- one with *ICMax* chip: **OK**

We have to find why there is a different behaviour between these two chips and how integrate your modification/update (at litedram phy level).

For RAM: size parameters are good and matchs RAM datasheet we have to investigate why a factor of two is introduces when size is computed.

Thanks!

enjoy-digital, 2023-08-25:
Thanks @trabucayre, @bacintom: Can you tell us which DDR3 variant is mounted on your Primer (SKHynix or ICMax)?

bacintom, 2023-08-25:
Hi!
Thanks for yout replies!
In my case, it is a SK Hynix. The mystery thickens.
I somehow managed to take a photo with a readable part number, let me attach it here.
[image]

bacintom, 2023-08-25:
@trabucayre to be honest I don't understand much about RAM but here's my take about the size issue:

I found a datasheet of H5TQ1Gx3EFR-xxx modules

In my case the exact number is H5TQ1G63EFR-PBC
the "63" (unlike "83") means it is a 64M x 16 configuration

There is a table called "Row and column address table"
[image]

The right column should be used
-There we have "row address" ranging from A0 to A12, I guess those are pin numbers... That's 13 bits, so the field 'nrows' of the DDR3Module should be 2^13 = 8192 (not 16384 as in the master)
-Column address is A0-A9 -> 10bits -> 2^10=1024 OK

What do you think?

e-yes, 2023-08-25:
Hi @bacintom could you try this (once again, not sure about proper timings):
https://github.com/enjoy-digital/litex/issues/1649#issuecomment-1656509253

bacintom, 2023-08-26:
@e-yes As long as I set nrows=8192 (the quickest way to test is, take an unmodified master and set the value in MT41J128M16) it will report the correct amount 128MiB, no other change needed.

So your change works and reports 128MiB. It will also fail the test in BIOS, and succeed after calling the sdram_mr_write/sdram_cal commands.

nickoe, 2023-08-27:
Thanks to the fix in from @trabucayre in  https://github.com/enjoy-digital/litedram/commit/6dadc11b21a8253aa20305f1a54c1a2f1f4bbfd6 I can make it pass and this configuration for the ddr3 module:

```
class H5TQ1G63EFR(DDR3Module):
    # geometry
    nbanks = 8
    nrows  = 8192
    ncols  = 1024
    # timings
    technology_timings = _TechnologyTimings(tREFI=64e6/8192, tWTR=(4, 7.5), tCCD=(4, None), tRRD=(4, 6), tZQCS=(64, 80))
    speedgrade_timings = {
        "800": _SpeedgradeTimings(tRP=15, tRCD=15, tWR=15, tRFC=(120, None), tFAW=(None, 50), tRAS=37.5),
        "1066": _SpeedgradeTimings(tRP=13.125, tRCD=13.125, tWR=15, tRFC=(160, None), tFAW=(None, 50), tRAS=37.5),
        "1600": _SpeedgradeTimings(tRP=13.75, tRCD=13.75, tWR=15, tRFC=(160, None), tFAW=(None, 40), tRAS=35),
    }
    speedgrade_timings["default"] = speedgrade_timings["800"]
```
Note the nrows as e-yes is also mentioning.

I am not sure how I got to those numbers as such, but some of them are either taken by searching for tRP numbers from other modules and probably https://github.com/enjoy-digital/litex/issues/1649#issuecomment-1656509253 -- but it is what I tested.

I get highest speed with the 800 timing, ... I don't know if that is expected.
```
  Write speed: 23.4MiB/s
   Read speed: 8.3MiB/s
```

For my 1066 and 1600 configration:
```
  Write speed: 22.2MiB/s
   Read speed: 7.8MiB/s
```

@bacintom after I got working sdram, the sdcard also worked for me. At least it would boot the demo.bin from a boot.json.

bacintom, 2023-08-28:
@nickoe thanks I hadn't noticed something was pushed to litedram master.

With @trabucayre 's commit, the bios passes the sdram check successfully. No need to tweak with mode register commands anymore and that's nice

If I use @nickoe 's timings, sdcard (non-spi) also works. Well, at least I can go as far as spisdcard: it shows "liftoff!" and freezes.

So, issues 1,2,3 are pretty much fixed. Thanks everyone!

verilogzhou, 2023-09-01:
@bacintom Regarding the liftoff issue, it can be seen from the following link that linux-on-litex-vexriscv supports vexriscv_smp,so it should be --cpu-type vexriscv_smp, but it will compile and prompt an error.
https://github.com/litex-hub/linux-on-litex-vexriscv

e-yes, 2023-09-02:
I have tried to compile the gateware for vexriscv_smp, but it doesn't fit and the compilation failed.

bacintom, 2023-09-03:
Hi,

I got over my liftoff issue. I did the following :

-update all litex. Patches about supporting tang primer 20k have been pushed recently.  

-check the files in the SD card using md5sum. One of my files had a wrong signature so I had to copy it again. I would recommend to double-check like this: unmount, unplug sd card, plug sd card, remount, verify checksum. This is to avoid being tricked by the OS, it could keep files in cache and skip reading from the actual SD card.
 
-building with linux-on-litex-vexriscv: `./make.py --board=sipeed_tang_primer_20k --toolchain=gowin --build --load` (a patch added sipeed_tang_primer_20k on August 29th)

With this the board can boot linux!

I could also build with video_terminal, but it's unstable. Only part of the text appears on screen and the board will freeze before linux is fully booted.

Sujunyou, 2023-09-09:
@bacintom hi , i firstly use litex to build tang-primer-20k linux , but in this linux , i find commands can not be uesd such as cd ,ls and so on. How do you know about these commands? Is there a corresponding instruction manual?

sususjysjy, 2023-09-10:
> 

hi , i use same  command:`./make.py --board=sipeed_tang_primer_20k --toolchain=gowin --build --load` ,but i got
same error like 4 :
[image]
how do you solve this problem

bacintom, 2023-09-10:
@Sujunyou cd and ls work fine in this linux, so I'm not sure what your problem is... 

Are you sure you set up a linux?... If you just build litex it will bring you to the litex bios, only a few bios-level commands are available.

There are some explanations here: https://github.com/litex-hub/linux-on-litex-vexriscv
in particular, you should download linux_2022_03_23.zip from https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164 , format an SD card with a FAT16 or FAT32 partition and extract the files there. However, about the file rv32.dtb I use the one built by make.py in the linux-on-litex-vexriscv/images because it has a different signature, I'm not sure whether it is important.

sususjysjy, 2023-09-10:
> @Sujunyou cd and ls work fine in this linux, so I'm not sure what your problem is...
> 
> Are you sure you set up a linux?... If you just build litex it will bring you to the litex bios, only a few bios-level commands are available.
> 
> There are some explanations here: https://github.com/litex-hub/linux-on-litex-vexriscv in particular, you should download linux_2022_03_23.zip from [litex-hub/linux-on-litex-vexriscv#164](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164) , format an SD card with a FAT16 or FAT32 partition and extract the files there. However, about the file rv32.dtb I use the one built by make.py in the linux-on-litex-vexriscv/images because it has a different signature, I'm not sure whether it is important.

thanks fo your reply.i follow your steps to build linux(use command:`python3 -m litex_boards.targets.sipeed_tang_primer_20k --with-sdcard --cpu-type vexriscv --cpu-variant linux --with-video-terminal --build --load
`)but i got following error:
`ERROR (RP0006) : The number(61480(56055 LUTs, 1537 ALUs, 0 ROM16s, 648 SSRAMs)) of logic in the design exceeds the resource limit(20736) of current device
ERROR (RP0001) : The number(76249) of DFF(DL) in the design exceeds the resource limit(16173) of current device(GW2A-LV18PG256C8/I7)
GowinSynthesis finish

    while executing
"run all"
    (file "run.tcl" line 11)
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/python-litex/litex-boards/litex_boards/targets/sipeed_tang_primer_20k.py", line 239, in <module>
    main()
  File "/root/python-litex/litex-boards/litex_boards/targets/sipeed_tang_primer_20k.py", line 228, in main
    builder.build(**parser.toolchain_argdict)
  File "/root/python-litex/litex/litex/soc/integration/builder.py", line 367, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/root/python-litex/litex/litex/soc/integration/soc.py", line 1332, in build
    return self.platform.build(self, *args, **kwargs)
  File "/root/python-litex/litex/litex/build/gowin/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/root/python-litex/litex/litex/build/generic_toolchain.py", line 123, in build
    self.run_script(script)
  File "/root/python-litex/litex/litex/build/gowin/gowin.py", line 150, in run_script
    raise OSError("Error occured during Gowin's script execution.")
OSError: Error occured during Gowin's script execution.
`it seems resources is not available.

bacintom, 2023-09-10:
@sususjysjy The issue 4 is not solved, I just disable the check that fails by editing gowin_gw1n.py, it's also fully explained in my first post...

bacintom, 2023-09-10:
@sususjysjy I'm confused, I was replying to @Sujunyou there, are you the same person?

I get the same error as you nowadays if I use litex_boards, the error means the design won't fit in the FPGA.
But it works if I use linux-on-litex-vexriscv/make.py instead... I don't know why 🤷

sususjysjy, 2023-09-10:
> @sususjysjy I'm confused, I was replying to @Sujunyou there, are you the same person?
> 
> I get the same error as you nowadays if I use litex_boards, the error means the design won't fit in the FPGA. But it works if I use linux-on-litex-vexriscv/make.py instead... I don't know why 🤷

> @sususjysjy I'm confused, I was replying to @Sujunyou there, are you the same person?
> 
> I get the same error as you nowadays if I use litex_boards, the error means the design won't fit in the FPGA. But it works if I use linux-on-litex-vexriscv/make.py instead... I don't know why 🤷

sorry,these all my account. I find reasons following:
lin itex,if use above command,generate all the sdcard;
but if you use  linux-on-litex-vexriscv , it generates spisdcard, it's slower than litex

bacintom, 2023-09-10:
@sususjysjy I can build with make.py and enable sdcard instead of spisdcard. It will build successfully. The size difference must be somewhere else?

But sdcard+linux-on-litex-vexriscv doesn't work on master with the tang primer 20k... I've just tried with @nickoe 's RAM timings, but I could not boot, sdcard_init fails again. 🤷🤷🤷

So, I think it's better to use spisdcard for now.

sususjysjy, 2023-09-10:
yeah,i use make.py, spi sdcard can be builed succeessfully ; but in litex, i use command `python3 -m litex_boards.targets.sipeed_tang_primer_20k --with-sdcard --build` , it will show me resources are not available。
i'm so confused for this

sususjysjy, 2023-09-11:
> @sususjysjy I can build with make.py and enable sdcard instead of spisdcard. It will build successfully. The size difference must be somewhere else?
>
> But sdcard+linux-on-litex-vexriscv doesn't work [...]

hi , you said "you can use make.py and enable sdcard",in make.py,it builds spisdcard , should i convert spisdcard to sdcard?
The following is the default version：
[image]
if i convert to sdcard , it will occurr error
May I take a look at your project？

sususjysjy, 2023-09-13:
> @sususjysjy what version of GOWIN IDE do you use?

thanks for your reply
i use the latest version 1.9.8.11_education
now i meet the new problem , i build .fs and load to tang-primer-20k , and bios work fine
but when i want to boot linux(provided by (linux-on-litex-vexriscv) from sdcard , terminal shows following error:
[image]
i don't knpw what's wrong

e-yes, 2023-09-13:
You are on the right path I believe. Can you give us more details?
I am even jealous, that I didn't come so close to solving this puzzle)

sususjysjy, 2023-09-13:
> You are on the right path I believe. Can you give us more details? I am even jealous, that I didn't come so close to solving this puzzle)

i use linux-on-litex-vexriscv , use the command ` ./make.py --board=sipeed_tang_primer_20k --toolchain=gowin --build` ,then i load the .fs to board .
and then , i put five files in sdcard ,and boot the board
finally , i see the following terminal:
[image]
[image]

e-yes, 2023-09-13:
Almost there
https://codebrowser.dev/linux/linux/drivers/soc/litex/litex_soc_ctrl.c.html#48

bentwire, 2023-09-14:
Ok this is amazing.  I had no idea this board had the FPGA resources to boot linux on!  Is the vexriscv core the only one to use for this?

e-yes, 2023-09-14:
I suspect vexriscv_smp is required at least for these prebuilt images.

sususjysjy, 2023-09-16:
i have solve this problem.thanks all

e-yes, 2023-09-17:
It would be nice if you share some details...

sususjysjy, 2023-09-28:
> It would be nice if you share some details...

ok. In a few days, I will post my strategy for how to solve the problem

bentwire, 2023-09-29:
I got it working on my board by ONLY replacing the image files with the ones from the download site.  If you overwrite the DTB that was created by litex when you build for this board all the hardware will be wrong.

Try without overwriting rv32.dtb and boot.json that the litex-on-linux-rexriscv build creates.

sususjysjy, 2023-10-02:
> I got it working on my board by ONLY replacing the image files with the ones from the download site. If you overwrite the DTB that was created by litex when you build for this board all the hardware [...]

you are right.:)
The same issues occurr in tang_mega_138k.
So confused about memory.(memory init fail)
I tried all the combinations of cl and cwl , but it works badly.
[image]
I think this is related to the timing issue caused by Gowin synthesis

BobMa0420, 2023-10-16:
> Hi,
>
> I got over my liftoff issue. I did the following :
>
> -update all litex. Patches about supporting tang primer 20k have been pushed recently.
>
> -check the files in the SD card using md5sum. One of [...]

Hi, I run `./make.py --board=sipeed_tang_primer_20k --toolchain=Gowin --with-video-terminal --build` and got an error:

> make.py: error: unrecognized arguments: --with-video-terminal

How can I build the bitstream with video_terminal? Thank you so much!

LeenOlll, 2023-10-16:
Hello,firstly I used the command  `./make.py --board=sipeed_tang_primer_20k --build` to generate bitstream file successfully. Then I used `./make.py --board=sipeed_tang_primer_20k --load` to it but it freezed like the picture for a long time. Is it normal?
[image]

gr8linux, 2023-10-23:
I find out that the big different between build up the litex over the make.py for the linux-on-litex and the way we build it up is to set the l2 cache if you like to have the same size and be able to build it up without error I could recommend to build up the litex with the below option:
python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512 --with-spi-sdcard --build --with-video-terminal
please update all of your packages with pip before try to run the above command.
To let the linux work as you expected the linux kernel also need to have the related driver its not enough to just add the HW like framebuffer and usb to the litex only.
The primer 20k is not a big fpga and as far as I find see its not capable of running many peripherals at the same time.
You need about 10K to 15K LUT for a soft CPU
Around 8k for HDMI 
UART 1 or 2 K
So the capacity is so tide for expansion.
A real deal on linux running board might be something like Arty A7 100K  you could run all you need and drive HD monitor

BobMa0420, 2023-10-24:
Hi, @gr8linux , thank you so much for your suggestions. I use `python3 -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512 --with-spi-sdcard --build --with-video-terminal` to generate bitstream file and load it on my board. Then I use the files of [https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164#issue-767463348](url) and try to boot Linux  from SDCard. However, it finally stuck in the "liftoff" phase. Is it a problem with the related driver as you mention above? Is there a way to slove that? Thanks!

[image]

gr8linux, 2023-10-27:
> Hi, @gr8linux , thank you so much for your suggestions. I use `python3 -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512 --with-spi-sdcard --build --with-video-terminal` to generate [...]

This might be related to the cpu version and LiteDRAM module maybe the below option help:

`python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512  --build --cpu-type vexriscv_smp --cpu-variant linux --build  --with-spi-sdcard --with-wishbone-memory`

I tried to remove the LiteDRAM module by --with-wishbone-memory and let it run as of SMP cpu model --cpu-type vexriscv_smp
The --cpu-variant linux will help to adjust the memory to load opensbi as needed to boot up the linux.

watermeko, 2024-10-11:
> This might be related to the cpu version and LiteDRAM module maybe the below option help:
>
> `python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512 --build --cpu-type vexriscv_smp [...]

That helps me a lot. Thank you. But could you please tell me what is the influence of "--with-wishbone-memory" ?

watermeko, 2024-10-11:
@gr8linux Hello. I follow your advice and run the command:
`python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512  --build --cpu-type vexriscv_smp --cpu-variant linux --build  --with-spi-sdcard --with-wishbone-memory`
But i encountered kernel panic when booting. Do you have any idea?

[    0.000000] Linux version 5.14.0 (florent@panda) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2021.08-381-g279167ee8d) 10.3.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Sep 21 12:57:31 CEST 2021
[    0.000000] Machine model: sim
[    0.000000] earlycon: liteuart0 at I/O port 0x0 (options '')
[    0.000000] Malformed early option 'console'
[    0.000000] earlycon: liteuart0 at MMIO 0xf0001000 (options '')
[    0.000000] printk: bootconsole [liteuart0] enabled
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x8
[    0.000000] SBI TIME extension detected
[    0.000000] SBI IPI extension detected
[    0.000000] SBI RFENCE extension detected
[    0.000000] SBI v0.2 HSM extension detected
[    0.000000] riscv: ISA extensions aimp
[    0.000000] riscv: ELF capabilities aim
[    0.000000] percpu: Embedded 8 pages/cpu s11340 r0 d21428 u32768
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=liteuart earlycon=liteuart,0xf0001000 rootwait root=/dev/ram0
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 48644K/65536K available (5685K kernel code, 572K rwdata, 883K rodata, 209K init, 221K bss, 16892K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 32 local interrupts mapped
[    0.000000] plic: interrupt-controller@f0c00000: mapped 32 interrupts with 1 handlers for 2 contexts.
[    0.000000] random: get_random_bytes called from start_kernel+0x4ac/0x63c with crng_init=0
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x171024e7e0, max_idle_ns: 440795205315 ns
[    0.000060] sched_clock: 64 bits at 100MHz, resolution 10ns, wraps every 4398046511100ns
[    0.008482] Console: colour dummy device 80x25
[    0.013375] Calibrating delay loop (skipped), value calculated using timer frequency.. 200.00 BogoMIPS (lpj=400000)
[    0.023232] pid_max: default: 32768 minimum: 301
[    0.035936] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.043097] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.124893] ASID allocator using 9 bits (512 entries)
[    0.136439] rcu: Hierarchical SRCU implementation.
[    0.160892] smp: Bringing up secondary CPUs ...
[    0.164792] smp: Brought up 1 node, 1 CPU
[    0.185935] devtmpfs: initialized
[    0.278793] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.288025] futex hash table entries: 256 (order: 2, 16384 bytes, linear)
[    0.320193] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.878961] pps_core: LinuxPPS API ver. 1 registered
[    0.882997] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.891759] PTP clock support registered
[    0.906587] FPGA manager framework
[    0.957288] clocksource: Switched to clocksource riscv_clocksource
[    1.574782] NET: Registered PF_INET protocol family
[    1.584683] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    1.609424] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    1.618002] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    1.625851] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    1.632652] TCP: Hash tables configured (established 1024 bind 1024)
[    1.642111] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    1.649295] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    1.741926] Unpacking initramfs...
[    1.865327] workingset: timestamp_bits=30 max_order=14 bucket_order=0
[    2.698446] io scheduler mq-deadline registered
[    2.702481] io scheduler kyber registered
[    3.896695] Kernel panic - not syncing: Scratch register read error - the system is probably broken! Expected: 0x12345678 but got: 0x0
[    3.905561] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.14.0 #1
[    3.909523] Hardware name: sim (DT)
[    3.911584] Call Trace:
[    3.913077] [<c0003b64>] dump_backtrace+0x2c/0x3c
[    3.916618] ---[ end Kernel panic - not syncing: Scratch register read error - the system is probably broken! Expected: 0x12345678 but got: 0x0 ]---

watermeko, 2024-10-11:
@gr8linux  Hello. I follow your advice and run the command:
`python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512  --build --cpu-type vexriscv_smp --cpu-variant linux --build  --with-spi-sdcard --with-wishbone-memory`
But i encountered kernel panic when booting. Do you have any idea?

[image]
[image]

TONYHEGUAGUA, 2024-10-17:
> @gr8linux Hello. I follow your advice and run the command: `python -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512 --build --cpu-type vexriscv_smp --cpu-variant linux --build [...]

I can boot linux successfully just by this command

VaibhavAttre, 2025-09-18:
I had an issue, which may / may not be related to the original post but this was the only thread I found relating to booting linux on the sipeed tang primer 20k. I followed the instructions provided in the linux-on-litex-vexriscv github page, and am assuming was able to build and flash the required files. 

Below is what my project directory consists of:

`vattre@vattre:~/fpga/linux-on-litex-vexriscv/build/sipeed_tang_primer_20k/gateware$ ls
impl  run.tcl                     sipeed_tang_primer_20k.fs        sipeed_tang_primer_20k_rom.init  sipeed_tang_primer_20k_sram.init  stream.cfg
prog  sipeed_tang_primer_20k.cst  sipeed_tang_primer_20k_mem.init  sipeed_tang_primer_20k.sdc       sipeed_tang_primer_20k.v
vattre@vattre:~/fpga/linux-on-litex-vexriscv/build/sipeed_tang_primer_20k/gateware$ cd ..
vattre@vattre:~/fpga/linux-on-litex-vexriscv/build/sipeed_tang_primer_20k$ cd ..
vattre@vattre:~/fpga/linux-on-litex-vexriscv/build$ cd ..
vattre@vattre:~/fpga/linux-on-litex-vexriscv$ ls
'='          build                HOWTO.md   LICENSE   prog          riscv64-unknown-elf-toolchain-10.2.0-2021.01.8-x86_64-linux-14.16.tar.gz   soc_linux.py   test
 adapter     build-gateware.log   images     make.py   __pycache__   set                                                                        st-link
 boards.py   buildroot            jlink      parport   README.md     sim.py                                                                     stream.cfg
vattre@vattre:~/fpga/linux-on-litex-vexriscv$ cd images
vattre@vattre:~/fpga/linux-on-litex-vexriscv/images$ ls
boot.json  boot_mmcblk0p2.json  boot_ram0.json  boot.vfat  build  Image  opensbi.bin  rootfs.cpio  rootfs.ext4  rv32.dtb  sdcard.img
vattre@vattre:~/fpga/linux-on-litex-vexriscv/images`

To build and load the files into the fpga, i run the following command:

`python3 -m litex_boards.targets.sipeed_tang_primer_20k --l2-size 512  --build --cpu-type vexriscv_smp --cpu-variant linux --build  --with-spi-sdcard --with-wishbone-memory --load`

And I get the following result (due to which I assume the files were loaded onto the fpga correctly)

`

INFO:SoC:Finalized SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode:  RW, Cached: False, Linker: False
Bus Regions: (7)
rom                 : Origin: 0x00000000, Size: 0x00020000, Mode:  RX, Cached:  True, Linker: False
sram                : Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False
main_ram            : Origin: 0x40000000, Size: 0x08000000, Mode: RWX, Cached:  True, Linker: False
opensbi             : Origin: 0x40f00000, Size: 0x00080000, Mode:  RW, Cached:  True, Linker:  True
csr                 : Origin: 0xf0000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False
clint               : Origin: 0xf0010000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False
plic                : Origin: 0xf0c00000, Size: 0x00400000, Mode:  RW, Cached: False, Linker: False
Bus Masters: (1)
- cpu_bus0
Bus Slaves: (6)
- plic
- clint
- rom
- sram
- main_ram
- csr
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
CSR Locations: (9)
- buttons        : 0
- ctrl           : 1
- uart           : 2
- timer0         : 3
- ddrphy         : 4
- identifier_mem : 5
- leds           : 6
- sdram          : 7
- spisdcard      : 8
INFO:SoC:IRQ Handler (up to 32 Locations).
IRQ Locations: (3)
- noirq  : 0
- uart   : 1
- timer0 : 2
INFO:SoC:--------------------------------------------------------------------------------
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ood_Wm_Hb1
...
openFPGALoader --cable ft2232 --bitstream /home/vattre/fpga/linux-on-litex-vexriscv/build/sipeed_tang_primer_20k/gateware/sipeed_tang_primer_20k.fs
empty
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file Parse /home/vattre/fpga/linux-on-litex-vexriscv/build/sipeed_tang_primer_20k/gateware/sipeed_tang_primer_20k.fs: 
Done
DONE
Load SRAM: [==================================================] 100.00%
Done
DONE`
 
However, when I finally run `litex_term --serial-boot --images images/boot.json /dev/ttyUSB0 --speed 115200` to boot linux, I am met with an empty line and a blinking cursor. Even when I let it run for more than an hour, I see nothing suggesting that the linux is booting or loading or anything of that sort. I am wondering if I skipped any step? I also cloned the buildroot repo to build the linux binaries. Help would be greatly appreciated!

enjoy-digital, 2026-10-21:
Closing since there was no further feedback after the update. Feel free to reopen if issues remain with current LiteX/linux-on-litex-vexriscv.

closed 2026-10-21 by enjoy-digital