enjoy-digital/litex #1766: Simulink-like design environment

davidbrochart, 2023-09-04:
There is a new Python-based tool called [Flojoy](https://www.flojoy.ai) that is similar to Simulink, which could be used to create digital hardware systems by connecting blocks together.
I'm wondering if something like that already exists for LiteX? Otherwise, what are your thoughts on integrating LiteX in Flojoy, as a no-code solution to create hardware designs?

josuah, 2023-09-04:
LiteX brings the ground foundation for building an SoC, on top of which gets possible to add more external physical hardware (plugging SPI/I2C/MIPI/... chips to it) and then adding custom cores as well.

Flojoy seems a lot more like something about data processing than bringing up an SoC together.

So it seems like integration would be more fit through some LiteX core that handles all the connection with a Flojoy design. If there is some configuration format that a Flojoy can generate, which specify all required inputs/outputs, then this could be fed into that custom core to generate the connection with LiteX.

This means that a bit of manual work would need to be done on the board of interest (instantiate the Flojoy core and connect it to the various resources of LiteX) but then the rest would happen in Flojoy and new flojoy updated design would
require modification.

Does this picture how you aimed to integrate it?

davidbrochart, 2023-09-04:
Not quite, I was rather thinking about something like [HDL Coder](https://www.mathworks.com/products/hdl-coder.html). Flojoy may look like a dataflow system at the moment, but it's still early stage and I think there is room for other use cases. For instance, hierarchical nodes would be a great fit to model hardware blocks.
But of course since a LiteX system could be defined in Flojoy, it could then interact with any other Flojoy component.

enjoy-digital, 2023-09-06:
Thanks @davidbrochart for sharing, I indeed wanted to explore solutions to easily connect LiteX's stream modules together, Flojoy could be interesting for this, I'll have a closer look.

mithro, 2023-09-06:
In a somewhat related note -- Antmicro was talking about a tool called "Visual System Designer" a few months back. See the following links;
 * https://riscv-europe.org/media/proceedings/plenary/2023-06-06-14h15-Michael-GIELDA-slides.pdf
 * https://www.youtube.com/watch?v=-Ts43wiQ640 and https://www.youtube.com/watch?v=6XtnTkCRrww
 * https://github.com/antmicro/visual-system-designer-resources