enjoy-digital/litex #1771: ERROR: Project file:///home/riscv/python-litex/migen has a 'pyproject.toml' and its build backend is missing the 'build_editable' hook. Since it does not have a 'setup.py' nor a 'setup.cfg', it cannot be installed in editable mode. Consider using a build backend that supports PEP 660.

Peng-Hongxi, 2023-09-07:
ERROR: Project file:///home/riscv/python-litex/migen has a 'pyproject.toml' and its build backend is missing the 'build_editable' hook. Since it does not have a 'setup.py' nor a 'setup.cfg', it cannot be installed in editable mode. Consider using a build backend that supports PEP 660.

jersey99, 2023-09-08:
@enjoy-digital Any update on how to fix this? It does break all my CI builds that install litex from scratch. Cheers!

jersey99, 2023-09-08:
Seems to be related to this: https://github.com/m-labs/migen/commit/2cfee3e0db6fdca9b5918686ea77c93252e7cebd

enjoy-digital, 2023-09-08:
@jersey99: https://github.com/enjoy-digital/litex/commit/6c30cb8695aa3f0760c627993031a1f4fa772416 should do it if you are using `litex_setup.py` in your CI.

jersey99, 2023-09-09:
Thanks @enjoy-digital, this works

closed 2024-01-12 by enjoy-digital