enjoy-digital/litex #1772: NameError: name 'Signal' is not defined

SunLee7, 2023-09-08:
i tryed ubuntu 22.04 with python3.10 and ubuntu 20.04 with python3.8.
when i try to build for my ac701, python always called a error:

sunlee@ubuntu:~$ python3 -m litex_boards.targets.xilinx_ac701 --help
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/sunlee/litex-boards/litex_boards/targets/xilinx_ac701.py", line 14, in <module>
    from litex.gen import *
  File "/home/sunlee/litex/litex/__init__.py", line 3, in <module>
    from litex.tools.litex_client import RemoteClient
  File "/home/sunlee/litex/litex/tools/litex_client.py", line 16, in <module>
    from litex.tools.remote.etherbone import EtherbonePacket, EtherboneRecord
  File "/home/sunlee/litex/litex/tools/remote/etherbone.py", line 11, in <module>
    from litex.soc.interconnect.packet import HeaderField, Header
  File "/home/sunlee/litex/litex/soc/interconnect/packet.py", line 15, in <module>
    from litex.gen import *
  File "/home/sunlee/litex/litex/gen/__init__.py", line 4, in <module>
    from litex.gen.signal import *
  File "/home/sunlee/litex/litex/gen/signal.py", line 11, in <module>
    class Open(Signal):
NameError: name 'Signal' is not defined

How can i fix it?

closed 2023-09-08 by SunLee7

exitrip, 2024-04-07:
Apologies for reviving this closed issue, but it's the only search hit for this problem.  How did you fix it?  I've tried reinstalling litex a bunch of different ways.

exitrip, 2024-04-07:
Huzzah!  I had to manually rebuild and reinstall the migen whl from the migen submodule.  `pip3 install .` from the migen src.
The module installed by `./litex_setup.y` was empty:
```
>>> inspect.getmembers(migen)
[('__doc__', None), ('__file__', None), ('__loader__', <_frozen_importlib_external.NamespaceLoader object at 0x7f027eeaa290>), ('__name__', 'migen'), ('__package__', 'migen'), ('__path__', _NamespacePath(['/home/engin/oss-cad-suite/litex/migen'])), ('__spec__', ModuleSpec(name='migen', loader=<_frozen_importlib_external.NamespaceLoader object at 0x7f027eeaa290>, submodule_search_locations=_NamespacePath(['/home/engin/oss-cad-suite/litex/migen'])))]
```

vs after the manually install:

```
>>> for name, obj in inspect.getmembers(migen):
...     if inspect.isclass(obj):
...             print(obj)
... 
<class 'migen.genlib.fsm.AnonymousState'>
<class 'migen.fhdl.structure.Array'>
<class 'migen.fhdl.structure.Constant'>
<class 'migen.fhdl.decorators.CEInserter'>
<class 'migen.fhdl.structure.Case'>
<class 'migen.fhdl.structure.Cat'>
<class 'migen.fhdl.structure.ClockDomain'>
<class 'migen.fhdl.decorators.ClockDomainsRenamer'>
<class 'migen.fhdl.structure.ClockSignal'>
<class 'migen.fhdl.structure.Constant'>
<class 'migen.fhdl.structure.DUID'>
<class 'migen.fhdl.structure.Display'>
<class 'migen.genlib.fsm.FSM'>
<class 'migen.fhdl.module.FinalizeError'>
<class 'migen.fhdl.structure.Finish'>
<class 'migen.fhdl.simplify.FullMemoryWE'>
<class 'migen.fhdl.structure.If'>
<class 'migen.fhdl.specials.Instance'>
<class 'migen.fhdl.specials.Memory'>
<class 'migen.fhdl.simplify.MemoryToArray'>
<class 'migen.fhdl.module.Module'>
<class 'migen.fhdl.decorators.ModuleTransformer'>
<class 'migen.genlib.fsm.NextState'>
<class 'migen.genlib.fsm.NextValue'>
<class 'migen.genlib.record.Record'>
<class 'migen.fhdl.structure.Replicate'>
<class 'migen.fhdl.decorators.ResetInserter'>
<class 'migen.fhdl.structure.ResetSignal'>
<class 'migen.fhdl.structure.Signal'>
<class 'migen.sim.core.Simulator'>
<class 'migen.fhdl.simplify.SplitMemory'>
<class 'migen.fhdl.specials.TSTriple'>
```

exitrip, 2024-04-16:
... but manually re-installing the migen whl breaks all of the litex install??
```
⦗OSS CAD Suite⦘ engin@engin-think:~/oss-cad-suite/litex$ python3 -m litex_boards.targets.sipeed_tang_nano_20k --build
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/engin/oss-cad-suite/litex/litex-boards/litex_boards/targets/sipeed_tang_nano_20k.py", line 15, in <module>
    from litex.build.io import DDROutput
ModuleNotFoundError: No module named 'litex.build.io'
```

```
>>> import litex
>>> for name, obj in inspect.getmembers(litex):
...     if inspect.isclass(obj):
...             print(obj)
... 
>>>
```

exitrip, 2024-04-16:
... until you also `cd litex; pip install .`

AndrewD, 2024-04-16:
You will likely have more success if you use litex_setup.py as documented in the [quickstart](https://github.com/enjoy-digital/litex?tab=readme-ov-file#quick-start-guide)

exitrip, 2024-04-17:
I'm describing what needed to be done AFTER following the quickstart exactly on Ubuntu 22.04 under WSL2.  Using the `litex_setup.py --init --install` does not build migen correctly.  See my first comment.

AndrewD, 2024-04-17:
Maybe try creating and activating a python venv before installing?

exitrip, 2024-04-17:
This all is inside the virtual env created by oss-cad-suite, python 3.11.6

faried, 2024-12-16:
I ran into this today.  This error occurs if you run `python3 -m litex_boards.targets.<whatever>` inside a directory that has litex's repos.

Here's how to reproduce the problem, using paths in my home directory (adjust as needed):
```
mkdir -p ~/lib ~/src/litex
python3 -m venv ~/lib/litex-env
source ~/lib/litex-env/bin/activate
cd ~/src/litex
# download litex_setup.py here and make it executable
./litex_setup.py --init --install --config full
# the following command will fail
python3 -m litex_boards.targets.sipeed_tang_nano_20k --help
```

Running that last command outside `~/src/litex` will work.