enjoy-digital/litex #1773: issue when boot on tang-primer-20k

sususjysjy, 2023-09-11:
hi , i want to boot linux in my boards , but when i launch my board , linux boot seems not work fine,issues are following:
`[    3.886195] Kernel panic - not syncing: Scratch register read error - the system is probably broken! Expected: 0x12345678 but got: 0x0 [    3.895050] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.14.0 #1 [    3.899034] Call Trace: [    3.900525] [<c0003b64>] dump_backtrace+0x2c/0x3c [    3.904063] ---[ end Kernel panic - not syncing: Scratch register read error - the system is probably broken! Expected: 0x12345678 but got: 0x0 ]`
but in ubuntu 20.04 , i use` ./sim.py` , i can boot this linux successfully ,i don't know what's wrong whis issue

closed 2023-09-15 by sususjysjy

watermeko, 2024-10-11:
i encountered the same problem. can you tell me how to solve it?

bubble-song, 2025-01-08:
Same problem here.

grant-olson, 2025-02-02:
When using https://github.com/litex-hub/linux-on-litex-vexriscv I get this error if I use the image they link to in the README.md. However if I copy the files from the ./image subdirectory that is actually in the git hub repo, it works.

XunNian11, 2025-03-12:
> When using https://github.com/litex-hub/linux-on-litex-vexriscv I get this error if I use the image they link to in the README.md. However if I copy the files from the ./image subdirectory that is actually in the git hub repo, it works.

I have encountered the same problem as you, but I did not find the correct file, could you please share it with me? Thank you

grant-olson, 2025-03-12:
Sorry this is the link: https://github.com/litex-hub/linux-on-litex-vexriscv?tab=readme-ov-file#-generating-the-linux-binaries-optional

XunNian11, 2025-03-14:
> Sorry this is the link: https://github.com/litex-hub/linux-on-litex-vexriscv?tab=readme-ov-file#-generating-the-linux-binaries-optional

Thank you! I replaced the rv32.dtb file with the generated one. it works

yuKILLING, 2025-03-20:
> > Sorry this is the link: https://github.com/litex-hub/linux-on-litex-vexriscv?tab=readme-ov-file#-generating-the-linux-binaries-optional
> 
> Thank you! I replaced the rv32.dtb file with the generated one. it works

hey, i encountered the same problem, i should make this? 

Generating the Linux binaries (optional)

$ git clone http://github.com/buildroot/buildroot
$ cd buildroot
$ make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
$ make

and the output put in the image folder?

XunNian11, 2025-03-20:
https://github.com/enjoy-digital/litex/issues/1750#issuecomment-1704067374
Yeah, and then I followed this

yuKILLING, 2025-03-26:
> [#1750 (comment)](https://github.com/enjoy-digital/litex/issues/1750#issuecomment-1704067374) Yeah, and then I followed this

hmmm, i build but no rv32.dtb file...

XunNian11, 2025-03-30:
> > [#1750 (comment)](https://github.com/enjoy-digital/litex/issues/1750#issuecomment-1704067374) Yeah, and then I followed this
> 
> hmmm, i build but no rv32.dtb file...

git clone http://github.com/buildroot/buildroot
cd buildroot
make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
Try this and rebuild it

alberto-grl, 2025-03-30:
> > [#1750 (comment)](https://github.com/enjoy-digital/litex/issues/1750#issuecomment-1704067374) Yeah, and then I followed this
> 
> hmmm, i build but no rv32.dtb file...

It is working for me. Please follow the notes here

https://github.com/alberto-grl/Sipeed-Tang-Gowin-FPGA/blob/main/README.md

Ask me if it's not clear.

Regards