enjoy-digital/litex #1774: Pynq Z1 Terminal

meyanik, 2023-09-12:
After using the target script for digilent pynq z1 board, i can see the led chase in my board but when i try to get bios running, i get no response from the card. is there anything i miss ?

trabucayre, 2023-09-13:
Could you be more precise?

Which CPU have you selected? By default a vexriscv is used.

Do you try to use on board **UART** interface?
For *Pynq* (and any others boards based on a *Zynq*) default **UART** interface is not available from PL (FPGA side): this one is connected to the PS (ARM CPU) interface.
So, if its not the case, you have to add an external USB<->UART connected to `PMODA:0` (tx) and `PMODA:1` (rx)

meyanik, 2023-09-13:
Sorry about amateur description of the problem. I used this command to generate and load the bitstream:
`python3 -m litex_boards.targets.digilent_pynq_z1 --build --load`
after loadig the bitstream i can see the chasing leds on the board but i cant connect to litex with litex_term. Is there any way to add external usb using migen. I tried using `add_uart` before but failed. I'm a total newbie to LiteX so sorry if questions are meaningless

trabucayre, 2023-09-13:
Ok.
This line uses the default SoC (*vexriscv*) and the PS part is unused.
A serial interface is already [defined](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/digilent_pynq_z1.py#L39) for this board so no need to add extra commands.

You have to connect an external usb<->uart peripheral and using corresponding `/dev/ttyUSBx` where **x** will be 3 if the onboard interface was plugged before adaptor (see `dmesg` for more informations).

meyanik, 2023-09-17:
Thank you. using TTL Uart converter over o PMODs did the work. Now i get the bios screen :)

closed 2023-09-17 by meyanik