enjoy-digital/litex #1792: Zephyr: DTS generation fail on vexriscv (target: xilinx_zcu102)

lianakoleva, 2023-09-29:
`./litex/litex/tools/litex_json2dts_zephyr.py --dts overlay.dts --config overlay.config csr.json` returns
```Generating overlay for: uart
Generating overlay for: timer0
Generating overlay for: ethmac
  dtsi key 'ethmac' not found, disable ethmac
Generating overlay for: spiflash
  dtsi key  not found, disable spiflash
Generating overlay for: sdcard_block2mem
  dtsi key  not found, disable sdcard_block2mem
Generating overlay for: sdcard_core
  dtsi key  not found, disable sdcard_core
Generating overlay for: sdcard_irq
  dtsi key  not found, disable sdcard_irq
Generating overlay for: sdcard_mem2block
  dtsi key  not found, disable sdcard_mem2block
Generating overlay for: sdcard_phy
  dtsi key  not found, disable sdcard_phy
Generating overlay for: i2c0
  dtsi key  not found, disable i2c0
Generating overlay for: i2s_rx
  dtsi key  not found, disable i2s_rx
Generating overlay for: i2s_tx
  dtsi key  not found, disable i2s_tx
Generating overlay for: mmcm
  dtsi key  not found, disable mmcm
Generating overlay for: main_ram
  dtsi key 'main_ram' not found, disable main_ram
Generating overlay for: identifier_mem
  dtsi key  not found, disable identifier_mem
No overlay handler for: ctrl at 0xf0000000
No overlay handler for: leds at 0xf0001000```

I am using command `python3 -m litex-boards.litex_boards.targets.xilinx_zcu102 --build --no-compile --cpu-type vexriscv --csr-json csr.json` to build vexriscv. The tutorial additionally specifies `--with-ethernet` but this is not on the list of recognized arguments.

AndrewD, 2023-10-16:
What is the failure? The command is rather verbose, but those messages are informative, not fatal 

What were you expecting? The dts and config files should have been generated

enjoy-digital, 2024-01-12:
Closing the issue since the issue is not clear and no feedback. @lianakoleva: Please provide more information if you still have questions.

closed 2024-01-12 by enjoy-digital