enjoy-digital/litex #1798: IOStandard missing for Efinix PLL clock reference

va7deo, 2023-10-11:
While trying to build a target for Efinix Ti180 I discovered that the IOStandard gets dropped if the signal is used for a PLL reference.
I defined the pin in _io as
`
("clk50", 0, Pins("K18"), IOStandard("3.3_V_LVCMOS")), 
`

then in the CRG

`
clk50 = platform.request("clk50")
self.pll = pll = TITANIUMPLL(platform)
pll.register_clkin(clk50, 50e6)
`

When the target is built, the resulting iface.py is missing any reference to that pin's IOStandard.  iface.py is then run and generates the peri.xml which is also missing it.  If the IO bank for the reference is set to 3.3V then there is a voltage mismatch and the build fails as it defaults to 1.8V

In  _build_iface_gpio found in efinix.py, the reference clock gets excluded and along with it the IOStandard.   To get my build working I appended a property which corrects the voltage mismatch.   I'm not good with Python so the code is very suspect.

`                elif isinstance(excluded_io, Signal):
                    if sig == excluded_io.name:
                        excluded = True
                        for other in others:
                            if isinstance(other,IOStandard):
                                self.additional_iface_commands.append('design.set_property("{sig}","IO_STANDARD","{name}")'.format(sig=sig,name=other.name))
`

Lastly, I don't know if I was using LiteX wrong and there's another way to set the IOStandard.

trabucayre, 2023-11-06:
Hi,
Efinix is a bit different to any others FPGA's vendors. Some of primitives (including PLL) are out of verilog and managed by the *interface designer*.
In this specific case the input clock isn't something at verilog top level but it's used directly by the PLL.
If you check *build/yourBoard/gateware/iface.py* you may found a line like:
```
design.gen_pll_ref_clock("pll0", pll_res="PLL_TL0", refclk_src="EXTERNAL", refclk_name="clk25", ext_refclk_no="0")
```
`refclk_name` isn't really relevant but `ext_refclk_no` inform about which PLL's input is used.

Which is your board? To check this behaviour.
Thanks

va7deo, 2023-11-06:
I'm using a custom Ti180 board that has 3.3V oscillators.   When the peri.xml file is created for the Efinity project, the property to set the IO standard would not get created causing a mismatch between the clock and the bank voltages.

design.set_property("clk","IO_STANDARD","1.8_V_LVCMOS")