enjoy-digital/litex #1801: OSError: Unable to find valid Meson build system, please install it with: - pip3 install meson.

PrabhanjanSangam, 2023-10-15:
As per the guide I followed all the steps,but when I typed the command : **$ litex_sim --cpu-type=vexriscv** 

I got the above error:
Traceback (most recent call last):
  File "/home/prabhanjan/.local/bin/litex_sim", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'litex_sim')())
  File "/home/prabhanjan/litex/litex/tools/litex_sim.py", line 476, in main
    builder.build(
  File "/home/prabhanjan/litex/litex/soc/integration/builder.py", line 337, in build
    self._check_meson()
  File "/home/prabhanjan/litex/litex/soc/integration/builder.py", line 258, in _check_meson
    raise OSError(msg)
OSError: Unable to find valid Meson build system, please install it with:
- pip3 install meson.

Even though I have meson version 1.2.2 .

Help required.

rohitmr619, 2023-10-22:
It likely is unable to find the installation, check where pip3 installs your user libraries to and add the folder to your PATH environment variable.

PrabhanjanSangam, 2023-11-02:
Thank you it helped.

closed 2023-11-02 by PrabhanjanSangam