enjoy-digital/litex #1802: NEORV32 Memory initialization failed on Arty A7-100T

vogma, 2023-10-15:
Hello there,

when executing following command in the litex-boards directory, the NEORV32 CPU Type is compiled and flashed to my Arty A7-100T:
`./digilent_arty.py --variant=a7-100 --cpu-type=neorv32 --build --load`

The LED banner is also working, but when i connect to it with `litex_term /dev/ttyUSB1`, the Memtest failed:

[image]

When using `--cpu-type=vexriscv` or `--cpu-type=picorv32`  there is no error and executables run fine.
Is there anything specific in the setup that has to be changed  so that the NEORV32 CPU can be used? In [this NEROV32 Issue](https://github.com/stnolting/neorv32/issues/115) the memory init seems to be working on a nexys video FPGA.

Thank You!

Greetings 
Marco

enjoy-digital, 2023-10-16:
Hello @vogma,

the NEORV32 wrapper has been updated recently and some changes have been done on the memory map IIRC. I'll try to do some tests and fix. Thanks for reporting.

Unike267, 2023-11-03:
Hello,

I can confirm the issue with an Arty A7-35T. I noted a difference between the specifications of the RAM and the set-up used by litex.

According to [digilent reference](https://digilent.com/reference/programmable-logic/arty-a7/start), the DDR3 is  256 MB @ 333 MHz (667 MT/s). However, the default litex set-up prints 256.0MiB 16-bit @ 800MT/s (CL-7 CWL-5). Is it possible that the memory overclocking make the failure?

bsbatusesli, 2025-07-29:
Hello, 

Despite the wrapper, the problem persists on Arty A7-100T. 

Has a solution been found?

Thanks.