enjoy-digital/litex #1803: Zephyr RTOS support for LiteX SoC with 64 bits CVA6 cpu

KevinQhv, 2023-10-20:
I try to add the board as described in the Zephyr documentation. But I get a lot of driver errors when compiling the example code. I have created my Device Tree in relation to the implemented drivers and in relation to the litex_vexriscv board and the [Thales cv32a6_zybo](https://github.com/ThalesGroup/riscv-zephyr/tree/main/boards/riscv/cv32a6_zybo) implementation. But, i get the following errors when compiling the application _hello_world_ with the command : `west build -b litex_cva6 samples/hello_world/ --pristine`

```
-- west build: making build dir /home/chipsnet/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/chipsnet/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.26.4
-- Found Python3: /home/chipsnet/zephyrproject/.venv/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter 
-- Cache files will be written to: /home/chipsnet/.cache/zephyr
-- Zephyr version: 3.4.0-rc2 (/home/chipsnet/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
-- Board: litex_cva6
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.0 (/usr/local/zephyr-sdk-0.16.0)
-- Found toolchain: zephyr 0.16.0 (/usr/local/zephyr-sdk-0.16.0)
-- Found Dtc: /usr/local/zephyr-sdk-0.16.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/chipsnet/zephyrproject/zephyr/boards/riscv/litex_cva6/litex_cva6.dts
-- Generated zephyr.dts: /home/chipsnet/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/chipsnet/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/chipsnet/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/chipsnet/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/chipsnet/zephyrproject/zephyr/boards/riscv/litex_cva6/litex_cva6_defconfig'
Merged configuration '/home/chipsnet/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/chipsnet/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/chipsnet/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
CMake Warning at /home/chipsnet/zephyrproject/zephyr/CMakeLists.txt:870 (message):
  No SOURCES given to Zephyr library: drivers__clock_control

  Excluding target from build.

CMake Warning at /home/chipsnet/zephyrproject/zephyr/CMakeLists.txt:870 (message):
  No SOURCES given to Zephyr library: drivers__gpio

  Excluding target from build.

-- Configuring done (6.3s)
-- Generating done (0.1s)
-- Build files have been written to: /home/chipsnet/zephyrproject/zephyr/build
-- west build: building application
[1/123] Preparing syscall dependency handling

[3/123] Generating include/generated/version.h
-- Zephyr version: 3.4.0-rc2 (/home/chipsnet/zephyrproject/zephyr), build: v3.4.0-rc2-161-g7f0aff9045bc
[113/123] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/chipsnet/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/chipsnet/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/rv64imac_zicsr_zifencei/lp64/medany"  -L/home/chipsnet/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcmodel=medany  -mabi=lp64  -march=rv64imac_zicsr_zifencei  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /home/chipsnet/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: warning: cannot find entry symbol __start; not setting start address
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/arch/arch/riscv/core/libarch__riscv__core.a(isr.S.obj): in function `.L0 ':
/home/chipsnet/zephyrproject/zephyr/arch/riscv/core/isr.S:164: undefined reference to `__soc_is_irq'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: /home/chipsnet/zephyrproject/zephyr/arch/riscv/core/isr.S:514: undefined reference to `__soc_handle_irq'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/arch/arch/riscv/core/libarch__riscv__core.a(prep_c.c.obj): in function `_PrepC':
/home/chipsnet/zephyrproject/zephyr/arch/riscv/core/prep_c.c:32: undefined reference to `soc_interrupt_init'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a(intc_plic.c.obj): in function `plic_init':
/home/chipsnet/zephyrproject/zephyr/drivers/interrupt_controller/intc_plic.c:242: undefined reference to `z_riscv_irq_priority_set'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: /home/chipsnet/zephyrproject/zephyr/drivers/interrupt_controller/intc_plic.c:249: undefined reference to `arch_irq_enable'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: zephyr/drivers/timer/libdrivers__timer.a(litex_timer.c.obj): in function `sys_clock_driver_init':
/home/chipsnet/zephyrproject/zephyr/drivers/timer/litex_timer.c:87: undefined reference to `z_riscv_irq_priority_set'
/usr/local/zephyr-sdk-0.16.0/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: /home/chipsnet/zephyrproject/zephyr/drivers/timer/litex_timer.c:89: undefined reference to `arch_irq_enable'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/chipsnet/zephyrproject/zephyr/build
```

I used to compilate the SoC by LiteX for [CVA6](https://github.com/litex-hub/pythondata-cpu-cva6) and i use the drivers LiteX for UART and Timer in Device tree file and i use the driver "sifive,plic-1.0.0" for interrupt controller.

```
- Board : Digilent Arty A7-100T
- LiteX 2022.08
- Zephyr v3.4.0
- zephyr-sdk 0.16.1
- CPU CVA6 64 bits
```

Can the litex drivers implemented in zephyr depot be used with all litex SoC ?

kgugala, 2023-10-23:
Hi @KevinQhv thanks for reporting this. I see that you already reported this some time ago in Zephyr (https://github.com/zephyrproject-rtos/zephyr/issues/59332), but it wasn’t really answered there. Please tag the RISC-V related issues with the `risc-v` tag, so that they popup on our dashboard. We (@antmicro) are maintaining RISC-V in Zephyr. We also added LiteX/RISC-V support to Zephyr some time ago.

When it comes to your questions:

* The linker errors you see are, most probably, caused by incorrect system configuration (some missing configs in the defconfig files) and probably some issues with the KConfig files from the newly added board
* CVA6 uses standard hlic, clint and plic so you should use the following drivers:

  * riscv,cpu-intc
  * sifive,clint0
  * sifive,plic-1.0.0

accordingly

Can you link to the code you’re trying to build (Zephyr repository with the board addons you mentioned). Once we can look at the code we can provide some feedback on what is missing.

I suppose we should move this discussion back to the Zephyr repository, as it is, in fact, a Zephyr question (not really a LiteX one).

KevinQhv, 2023-10-27:
This issue is closed. It continues with the collaborators in the depot zephyr #https://github.com/zephyrproject-rtos/zephyr/issues/64350

closed 2023-10-27 by KevinQhv