enjoy-digital/litex #1805: How to drive `soc_rst` from an external pin?

shareefj, 2023-10-21:
Hi there,

I'm just starting to use Litex/Migen for the first time and am struggling to get my head around some of the concepts.  A first note: there doesn't seem to be any documentation on the clocking/reset strategy and nothing about some of the assumptions made about clock domain and reset naming.  For example, the fact that any `CRG` block should have an internal signal named `rst` that will be driven from the `SoCController`.

So after looking through the code, it seems to me that the `SoCController` is purely there to allow software to reset either the CPU or the full SoC.  If a `CRG` module has been instantiated with an attribute name of `crg`, then at the SoC level you drive `crg.rst` from `ctrl.soc_rst`.

But if I have an external push button reset term, what's the best way of driving `ctrl.soc_rst` from this?  I seem to have managed to hack the following top level together which sort of seems to generate something sensible but would like some feedback on the best way to hook up the resets.

Secondly, in the below code, if I add the `cd_sys` domain reset synchroniser to the `specials` list I get a multiply driven reset.  All the examples I've seen show this style so I'm obviously doing something else wrong here.

```python
import argparse
import logging

from migen import *
from migen.genlib.resetsync import AsyncResetSynchronizer

from litex.soc.integration.builder import Builder
from litex.soc.integration.soc_core import SoCCore
from litex.soc.interconnect.csr import AutoCSR, CSRStorage
from litex.soc.cores.clock import ECP5PLL
from litex_boards.platforms import lattice_versa_ecp5

logger = logging.getLogger(__name__)

class CRG(Module, AutoCSR):
    """Clock and Reset Generation"""

    def __init__(self, platform):
        self.rst = Signal()
        self.clock_domains.cd_sys = ClockDomain()
        self.clock_domains.cd_por = ClockDomain()

        clk_in = platform.request(platform.default_clk_name)
        clk_in_freq = round( 1e9 / platform.default_clk_period )

        # Push button reset
        self.rst_n = platform.request("rst_n")
        self.specials += AsyncResetSynchronizer(self.cd_por, ~self.rst_n)

        # Power on reset
        por_count = Signal(16, reset=2**16-1)
        por_done  = Signal()
        self.comb += self.cd_por.clk.eq(clk_in)
        self.comb += por_done.eq(por_count == 0)
        self.sync.por += If(~por_done, por_count.eq(por_count - 1))

        # PLL
        self.submodules.pll = pll = ECP5PLL()
        self.comb += pll.reset.eq(~por_done | self.rst)
        pll.register_clkin(clk_in, clk_in_freq)
        pll.create_clkout(self.cd_sys, 50e6, margin = 0)

        # self.specials += AsyncResetSynchronizer(self.cd_sys, ~pll.locked)
        AsyncResetSynchronizer(self.cd_sys, ~pll.locked)

class VersaSoC(SoCCore):
    """Top level of the Versa platform"""

    def __init__(self, **kwargs):
        platform = lattice_versa_ecp5.Platform(device="LFE5UM", toolchain="trellis")
        self.submodules.crg = CRG(platform)

        SoCCore.__init__(
            self, 
            platform=platform,
            clk_freq=50e6, 
            cpu_reset_address=0x10000000,
            **kwargs
        )

def get_parser():
    parser = argparse.ArgumentParser()
    parser.add_argument(
        "--verbose",
        action="store_true",
        help="Be verbose"
    )
    return parser

def main():
    parser = get_parser()
    args = parser.parse_args()

    level = logging.DEBUG if args.verbose else logging.INFO
    logging.basicConfig(level=level)

    logger.debug("Running in DEBUG mode")

    soc = VersaSoC(cpu_type="vexriscv", cpu_variant="minimal")
    builder = Builder(soc)
    builder.build()

    
if __name__ == "__main__":
    main()
```

shareefj, 2023-10-21:
Nothing like a bit of printf debugging - so my multi-driven net was due to the fact that calling `pll.create_clkout` also adds an async reset synchroniser to the same domain.  So having removed the duplicate it now seems to at least synthesise correctly.

So my only outstanding question is how would I go about assigning an external signal to either `soc_rst` or `cpu_rst` individually?

hvegh, 2023-10-22:
Something like:

```
        # Define output pins
        from litex.build.generic_platform import Pins, IOStandard
        platform.add_extension([
            ("reset2",  0, Pins("69"), IOStandard("LVCMOS33")),
        ])
    
        rst2 = platform.request("reset2"), 
```

shareefj, 2023-10-22:
Thanks but perhaps a better description is in order.

Inside of `litex.soc.integration.soc` there is a `SoCController` class that instantiates two CSRs that enable software to reset either just the CPU or the full SoC.  https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L792

Within the same file, in the `SoC` class, the CPU reset is driven by OR'ing these two signals together.  https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1143

And finally, the `soc_rst` is used to reset the `CRG` if it has an attribute named `rst`:  https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1181

So we effectively have the following RTL:

```verilog
assign soc_rst = soc_controller.soc_rst;
assign cpu_rst = soc_controller.cpu_rst;

assign cpu.reset = soc_rst || cpu_rst;
assign crg.rst = soc_rst;
```

What I want to do is add an external reset source to either the `soc_rst` or `cpu_rst` terms.  So let's say I grab that input pin using the usual `platform.request()` call, how do I feed that into those reset terms so that I end up with:

```verilog
assign cpu.reset = soc_rst || cpu_reset || external_reset;
```

shareefj, 2023-10-30:
OK, after a few questions on Discord, it seems that the way to do this is to work around the limitations of Migen by creating a new clock domain into which you synchronise the external reset.  Then you have a reset term that you can feed into the CPU reset.

```python
class CRG(Module, AutoCSR):
    """Clock and Reset Generation"""

    def __init__(self, platform):
        self.rst = Signal()
        self.clock_domains.cd_sys = ClockDomain()
        self.clock_domains.cd_por = ClockDomain(reset_less=True)
        self.clock_domains.cd_ext = ClockDomain()

        clk_in = platform.request(platform.default_clk_name)
        clk_in_freq = round(1e9 / platform.default_clk_period)

        # Power on reset
        por_count = Signal(16, reset=2**16 - 1)
        por_done = Signal()
        self.comb += self.cd_por.clk.eq(clk_in)
        self.comb += por_done.eq(por_count == 0)
        self.sync.por += If(~por_done, por_count.eq(por_count - 1))

        # PLL
        self.submodules.pll = pll = ECP5PLL()
        self.comb += pll.reset.eq(~por_done | self.rst)
        pll.register_clkin(clk_in, clk_in_freq)
        pll.create_clkout(self.cd_sys, 50e6, margin=0, with_reset=False)

        # The SYS domain is reset whenever the PLL is unlocked
        sys_rst = Signal()
        self.comb += sys_rst.eq(~pll.locked)
        self.specials += AsyncResetSynchronizer(self.cd_sys, sys_rst)

        # External push button reset synchronised into its own domain
        rst_n = platform.request("rst_n")
        self.comb += self.cd_ext.clk.eq(self.cd_sys.clk)
        self.specials += AsyncResetSynchronizer(self.cd_ext, ~rst_n)

class VersaSoC(SoCCore):
    """Top level of the Versa platform"""

    def __init__(self, **kwargs):
        platform = lattice_versa_ecp5.Platform(device="LFE5UM", toolchain="trellis")
        SoCCore.__init__(self, platform=platform, clk_freq=50e6, **kwargs)

        # Clock/Reset Generator
        self.crg = CRG(platform)

        # Hook up the push button reset
        self.comb += If(self.crg.cd_ext.rst, self.cpu.reset.eq(1))
```

Be careful of trying to do what you might expect is valid by just OR'ing in the reset term.  Migen uses non-blocking assignments for everything (!) so the following doesn't work:

```python
        # Hook up the push button reset
        self.comb += self.cpu.reset.eq(self.cpu.reset | ~self.crg.rst_n)
```

I think it's probably easier to write your CRG in SV and import that if you are doing anything more complex with clocks/resets.

closed 2023-10-30 by shareefj