enjoy-digital/litex #1807: RFC: register new Interrupt Service Routine (ISR) funtions for additional interrupts

hvegh, 2023-10-22:
Dear Litex community,

Currently the [isr.c](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/isr.c) only handles the uart.

**Question:** how to handle additional interrupts in software? I don't see currently an generic way of extending this.  

A possible API might look like:
```
int set_handler(unsigned int irq, void (*handler)(void));

void (*handler)(void) get_handler(unsigned int irq);

int remove_handler(unsigned int irq);
```

My hack for the moment is a callback function in the Litex isr routine shown below.

_My little hack to extend the isr routine:_
```
diff --git a/litex/soc/software/libbase/isr.c b/litex/soc/software/libbase/isr.c
index ac7a7920..898eb40a 100644
--- a/litex/soc/software/libbase/isr.c
+++ b/litex/soc/software/libbase/isr.c
@@ -191,6 +191,9 @@ void isr(void)
 }
 
 #else
+
+void (*isr_next)(unsigned int irqs) = NULL;
+
 void isr(void)
 {
        __attribute__((unused)) unsigned int irqs;
@@ -203,6 +206,8 @@ void isr(void)
                uart_isr();
 #endif
 #endif
+       if(isr_next)
+               (*isr_next)(irqs);
 }
 #endif
```

_Code snippet using the callback function for my pps core:_
```
extern void (*isr_next)(unsigned int irqs);
    
void isr_pps(unsigned int irqs)
{
  if (irqs & (1 << PPS0_INTERRUPT)) {
     // handle stuff
  }
}

static void pps_init(void) {
  isr_next = &isr_pps;
  irq_setmask(irq_getmask() | (1 << PPS0_INTERRUPT));
  pps0_ev_enable_write(1);
}
```

AndrewD, 2023-10-23:
There is another pr for a different approach to this.
I've got a more traditional irq handler registration (proof of concept) in our tree I've been meaning to push for discussion.

hvegh, 2023-11-10:
Now made obsolete by the pull request Andrew is working on. https://github.com/enjoy-digital/litex/pull/1815
Thanks!

closed 2023-11-10 by hvegh