enjoy-digital/litex #1813: sys_rst timings on large SoC

Dolu1990, 2023-10-25:
Hi,

Deploying a design which is ~ 60k lut on Digilent nexys video, the sys_rst timings are getting worst and worst.
Seems like using the FPGA regular interconnect to wire sys_rst everywere isn't good enough to reach 100 Mhz.

I manualy edited the generated verilog to have a BUFG on this path and it seems to solve the timing issue well enough.

Any idea how to implement that BUFG in litex ?

Thanks

enjoy-digital, 2023-10-26:
Hi @Dolu1990,

this should do it:

```patch
diff --git a/litex/build/xilinx/common.py b/litex/build/xilinx/common.py
index fb40cd3d..a438a253 100644
--- a/litex/build/xilinx/common.py
+++ b/litex/build/xilinx/common.py
@@ -71,7 +71,7 @@ class XilinxAsyncResetSynchronizerImpl(Module):
         if not hasattr(async_reset, "attr"):
             i, async_reset = async_reset, Signal()
             self.comb += async_reset.eq(i)
-        rst_meta = Signal()
+        rst_meta  = Signal(2)
         self.specials += [
             Instance("FDPE",
                 attr   = {"async_reg", "ars_ff1"},
@@ -80,7 +80,7 @@ class XilinxAsyncResetSynchronizerImpl(Module):
                 i_CE   = 1,
                 i_C    = cd.clk,
                 i_D    = 0,
-                o_Q    = rst_meta,
+                o_Q    = rst_meta[0],
             ),
             Instance("FDPE",
                 attr   = {"async_reg", "ars_ff2"},
@@ -88,8 +88,12 @@ class XilinxAsyncResetSynchronizerImpl(Module):
                 i_PRE  = async_reset,
                 i_CE   = 1,
                 i_C    = cd.clk,
-                i_D    = rst_meta,
-                o_Q    = cd.rst
+                i_D    = rst_meta[0],
+                o_Q    = rst_meta[1],
+            ),
+            Instance("BUFG",
+                i_I = rst_meta[1],
+                o_O = cd.rst,
             )
         ]
```

but will add it on each AsyncResetSynchronizer instance. If this works, I could try to only enable it on some specific clock domains (ex sys_clk).

enjoy-digital, 2023-11-07:
@Dolu1990: This should now be implemented with:
- https://github.com/enjoy-digital/litex/commit/c1e4b3a850ce84c039f09e360ed039a1d47ad936
- https://github.com/litex-hub/litex-boards/commit/bc74eeca7812b35b9edb02d67bb186caf55b9689

For now I've just used it on the Nexys Video target but we could apply it to other Xilinx boards once validated.

Dolu1990, 2023-11-07:
Nice thanks :D

Dolu1990, 2023-11-14:
Works well, thanks.

closed 2023-11-14 by Dolu1990

enjoy-digital, 2023-11-14:
Great, thanks for the feedback.