enjoy-digital/litex #1820: Physical constraint problem on tang_primer_20k

mmx-114514, 2023-10-28:
When I try to enable both hdmi and etnerbone on tang_primer_20k, the generated .cst physical constraint file has inconsistent voltages in the same bank: Bank 0 vccio(3.3) is locked by other constraint or embedded port, conflicting BANK_VCCIO set by 'hdmi_clk_p_obuf' : IO_TYPE = LVCMOS18 in the same bank. port, conflicting BANK_VCCIO set by 'hdmi_clk_p_obuf' : IO_TYPE = LVCMOS18 in the same bank How to solve this? Thank you for your help.

hvegh, 2023-10-29:
Yes... all the hdmi pins in the dock are pulled up using an external resistor:

Drop the  IOStandard("LVCMOS18") on these pins:

        Subsignal("hdp", Pins("CARD1:154"), IOStandard("LVCMOS18")),
        Subsignal("cec", Pins("CARD1:152"), IOStandard("LVCMOS18")),
        Subsignal("sda", Pins("CARD1:95"),  IOStandard("LVCMOS18")),
        Subsignal("scl", Pins("CARD1:97"),  IOStandard("LVCMOS18")),

enjoy-digital, 2023-11-09:
Hi @mmx-114514,

Can you share your build command to reproduce the issue? Thanks.

mmx-114514, 2023-11-09:
I used the way in linux-on-litex-vexriscv to generate vexriscv-smp via make.py, I used "python3 . /make.py --board=Sipeed_tang_primer_20k --with-coherent-dma --build" to generate the coherent DMA, and also added framebuffer and with video terminal. currently my screen displays the terminal information correctly. But I can find by coremark that the cpu score is much lower than without hdmi, I think this is caused by not being able to call video dma. Also in the generated dts file I can't find any configuration related to video with hdmi.
Thanks.

mmx-114514, 2023-11-09:
I Drop the IOStandard("LVCMOS18") on these pins.
    Subsignal("hdp", Pins("CARD1:154"), #IOStandard("LVCMOS18")),
    Subsignal("cec", Pins("CARD1:152"), #IOStandard("LVCMOS18")),
    Subsignal("sda", Pins("CARD1:95"),  #IOStandard("LVCMOS18")),
    Subsignal("scl", Pins("CARD1:97"),  #IOStandard("LVCMOS18")),