enjoy-digital/litex #1830: Broken QMtech Altera boards

DFL1973, 2023-11-08:
Hello, I'm very new on Litex and I got some errors building the Mistex using QMTech Altera. Building using a QMtech Xilinx board it´s working fine. The error seems to be on the Litex side, but sorry if I'm wrong. 

Traceback (most recent call last):
  File "/home/fpga/Mistex/MiSTeX-boards/mistex_boards/qmtech_5cefa5_daughterboard.py", line 188, in <module>
    handle_main(main)
  File "/home/fpga/Mistex/MiSTeX-boards/mistex_boards/util.py", line 81, in handle_main
    main(core=sys.argv[1])
  File "/home/fpga/Mistex/MiSTeX-boards/mistex_boards/qmtech_5cefa5_daughterboard.py", line 183, in main
    platform.build(Top(platform),
  File "/home/fpga/Mistex/Litex/litex/litex/build/altera/platform.py", line 45, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/fpga/Mistex/Litex/litex/litex/build/generic_toolchain.py", line 86, in build
    v_output = platform.get_verilog(self.fragment, name=build_name, **kwargs)
  File "/home/fpga/Mistex/Litex/litex/litex/build/altera/platform.py", line 38, in get_verilog
    return GenericPlatform.get_verilog(self, *args,
  File "/home/fpga/Mistex/Litex/litex/litex/build/generic_platform.py", line 469, in get_verilog
    return verilog.convert(fragment, platform=self, **kwargs)
  File "/home/fpga/Mistex/Litex/litex/litex/gen/fhdl/verilog.py", line 656, in convert
    verilog += _generate_hierarchy(top=LiteXContext.top)
  File "/home/fpga/Mistex/Litex/litex/litex/gen/fhdl/verilog.py", line 95, in _generate_hierarchy
    for l in hierarchy_explorer.get_hierarchy().split("\n"):
  File "/home/fpga/Mistex/Litex/litex/litex/gen/fhdl/hierarchy.py", line 58, in get_hierarchy
    r += self.get_tree(self.top)
  File "/home/fpga/Mistex/Litex/litex/litex/gen/fhdl/hierarchy.py", line 31, in get_tree
    for name, mod in module._submodules:
AttributeError: 'NoneType' object has no attribute '_submodules'

enjoy-digital, 2023-11-08:
Hi @DFL1973,

thanks for the feedback. I've been able to reproduce when trying to build MisteX and the build issue should be fixed with https://github.com/enjoy-digital/litex/commit/46107137975850b895e8d7cd56040898d1ec060b.

DFL1973, 2023-11-08:
It's working, confirmed. Thanks for the quick reply!

closed 2023-11-08 by DFL1973