enjoy-digital/litex #1831: PR #1817 may have broken AXI-lite peripheral interconnect

Dolu1990, 2023-11-08:
Hi,

I update Litex and tested NaxRiscv, but it seems that #1817 did break something related to the peripheral interconnect.

UART work enough for the bootloader, but both spi and mmc are broken (can't run the sdcard bootloader).
I tracked it down to #1817. 
The previous main branch commit works, while it seems that everything since #1817 isn't.

To visualize the issue, i did run :
- litex_sim --cpu-type=naxriscv  --with-sdram --sdram-data-width=64 --bus-standard axi-lite  --with-coherent-dma --xlen=32 --with-sdcard --trace --trace-fst

But the issue is that i do not know which signals represent the memory bus which goes to the sdcard controller.

Any idea ^^ ?
Thanks

enjoy-digital, 2023-11-08:
Thanks for the feedback @Dolu1990, I'll look at it.

enjoy-digital, 2023-11-08:
@Dolu1990: I started looking at it and can see the issue by directly comparing the generated verilog with/without the changes. I have to stop for now, will continue later or tomorrow.

Dolu1990, 2023-11-08:
Great, thanks :)

enjoy-digital, 2023-11-09:
This should be fixed with upstream.

Dolu1990, 2023-11-09:
Thanks, will try :)

Dolu1990, 2023-11-27:
Hi,

I forgot about closing that issue, but it seems it doesn't fix it all.
MMC work fine with linux, but ethernet is broken.
https://github.com/SpinalHDL/NaxRiscv/issues/69#issuecomment-1827751420
Likely some byte access done to the peripheral area by the driver or something like it ?

What were the fixes which were pushed to #1817 ?

enjoy-digital, 2023-11-28:
Hi @Dolu1990,

thanks for the feedback, I'll look at it. Would you mind sharing your build command for the test on hardware? I should be able to identify the issue by directly looking at the verilog before and after the PR as I did for the MMC.

Dolu1990, 2023-11-28:
Thanks ^^
I'm currently using : 
python3 -m litex_boards.targets.digilent_nexys_video --cpu-type=naxriscv  --bus-standard axi-lite --with-video-framebuffer --with-coherent-dma --with-sdcard --with-ethernet --xlen=64 --scala-args='rvc=true,rvf=true,rvd=true,alu-count=2,decode-count=2'  --with-jtag-tap --sys-clk-freq 100000000 --cpu-count 2 --build

enjoy-digital, 2023-12-08:
@Dolu1990: Sorry for the delay, I'm looking at it now. I've been able to have a simple repro, just noting it here if I need it later:

Netboot working:
`python3 -m litex_boards.targets.digilent_arty --bus-standard=wishbone --with-ethernet --build  --load`

Netboot failing:
`python3 -m litex_boards.targets.digilent_arty --bus-standard=axi-lite --with-ethernet --build  --load`

So Ethernet currently seems to be broken when used with axi-lite interface. I'll try to fix this now.

Dolu1990, 2023-12-08:
No worries :)
 Thanks

Dolu1990, 2023-12-15:
I just tested, it works all good now :D
I may also have fixed some weird boot freeze that i had by the past, didn't seems to happen anymore so far

Thanks

closed 2023-12-15 by Dolu1990

enjoy-digital, 2023-12-15:
Great, thanks for the feedback and test!