enjoy-digital/litex #1834: json2dts issue

Renemiess, 2023-11-09:
Hi, I wanna know how to convert json to dts, thanks

Renemiess, 2023-11-09:
when I run make.py or use this command("litex_json2dts_linux csr.json > rv32.dts"), I got this error below
[image]

AndrewD, 2023-11-10:
Which cpu did you build your target with?

What do you get if you use make.py to automate the linux-on-litex-vexriscv build?

Renemiess, 2023-11-10:
> Which cpu did you build your target with?
> 
> What do you get if you use make.py to automate the linux-on-litex-vexriscv build?

cpu: kianv, and board is sipeed_tang_primer_20k, and I don't know how to boot linux on this board

AndrewD, 2023-11-10:
Follow the instructions here:
https://github.com/litex-hub/linux-on-litex-vexriscv. This is a supported board:

$ ./make.py --board=sipeed_tang_primer_20k --build

There are a few other steps, but once you have the default, supported, setup working you can then start to explore other cpus, etc.