enjoy-digital/litex #1835: Thoughts on skipping interrups problem, on gowin / vexrisc platform

hvegh, 2023-11-10:
Sometimes my design is skipping interrupts:

I have added an extension core to the basic `sipeed_tang_nano_9k.py` it gets an 1Hz PPS pulse from a GPS module and it then measures the reference clock phase.  This is working pretty well however once every 5-15 seconds one ore more interrupts fail to trigger.  

Here is the setup:
```
           +--------------+
1Hz pps -->|              |<-- refclock (now system clock of 27MHz)
           |              |
           +--------------+
                |
               \/
          uart phase measurments         
```

Litex code:
```
...
        self.ev     = EventManager()
        self.ev.pps = EventSourceProcess(edge="rising")
        self.ev.finalize()

        self.comb += self.ev.pps.trigger.eq(self.i_pps)
...
        # add phase detector with system clock
        pps = platform.request("pps0")
        platform.add_period_constraint(pps, 1e9/10e6)

        self.add_module(name="pps0", module=
            PPS(platform, width=14, i_pps=pps)
        )
        if self.irq.enabled:
            self.irq.add("pps0", use_loc_if_exists=True)
...
```

And then in RISC-V we have:
```
void isr_pps(unsigned int irqs) {
  unsigned int stat;

  if (irqs & (1 << PPS0_INTERRUPT)) {
    stat = pps0_ev_pending_read();
    if (stat & 1) {
      int phase = pps0_phase_read();
      pps0_ev_pending_write(1);

      printf("pps0 phase = %d Hz\n", phase);
    }
  }
}

void pps_init(void) {
  isr_next = &isr_pps; // my temporary hack see https://github.com/enjoy-digital/litex/issues/1807
  pps0_ev_pending_write(pps0_ev_pending_read());
  pps0_ev_enable_write(1);
  irq_setmask(irq_getmask() | (1 << PPS0_INTERRUPT));
}
```
Why we are losing interrupts. 
- The interrupt is only once every second. 
- The pps pulse is clean as a whistle fast rise times 20% on/ 80% off duty cycle. Checked with scope.  
- The uart has ample time to print out the phase string...
I am puzzled...

hvegh, 2023-11-10:
Here is the some output, if there is an `*` then the isr routine got triggered.
```
*pps0  3798 -61 Hz
*pps0  2905 -61 Hz
*pps0  2011 -62 Hz
*pps0  1118 -61 Hz
*pps0   225 -61 Hz
*pps0 15716 -61 Hz
*pps0 14823 -61 Hz
*pps0 13930 -61 Hz
pps0 13037 -61 Hz
pps0 12143 -62 Hz
*pps0 11250 -61 Hz
*pps0 10357 -61 Hz
*pps0  9464 -61 Hz
pps0  8570 -62 Hz
*pps0  7677 -61 Hz
*pps0  6784 -61 Hz
*pps0  5891 -61 Hz
*pps0  4998 -61 Hz
*pps0  4105 -61 Hz
*pps0  3212 -61 Hz
*pps0  2318 -62 Hz
*pps0  1425 -61 Hz
pps0   533 -60 Hz
*pps0 16024 -61 Hz
*pps0 15130 -62 Hz
*pps0 14237 -61 Hz
*pps0 13344 -61 Hz
*pps0 12451 -61 Hz
*pps0 11558 -61 Hz
*pps0 10665 -61 Hz
*pps0  9772 -61 Hz

```

AndrewD, 2023-11-11:
My thoughts from the fragments you have posted: it look like you are directly using an  IO as the source for an edge triggered interrupt.

I suspect you will get far better results as follows:

from migen.genlib.cdc import MultiReg

pps = Signal()
self.specials += MultiReg(self.i_pps, pps)
self.comb += self.ev.pps.trigger.eq(pps)

hvegh, 2023-11-12:
Yes this did the trick, interrupts are now 100% reliable!
Thank you!

hvegh, 2023-11-12:
This fixed it for me

closed 2023-11-12 by hvegh