enjoy-digital/litex #1846: netboot

54o8, 2023-11-22:
hi 
What do I need to do if I use netboot?（tang_primer_20k）
I followed the tutorial to place the files in the image directory into the tfpt directory
[image]
but netboot failed
[image]
Do I need to make any changes to the local ip or remote ip

enjoy-digital, 2023-12-19:
Hi @54o8,

the two IP addresses seems to be on a different subnets, can you first do a test on the same subnet? or test your TFTP server from the same subnet than the FPGA board?

enjoy-digital, 2024-01-12:
Closing since no feedback. @54o8 if you still have trouble, please add following questions to this issue, we'll re-open.

closed 2024-01-12 by enjoy-digital