enjoy-digital/litex #1848: Tang Primer 20k dram issues

Glowman554, 2023-12-04:
Hey, i used the following command `python3 -m litex_boards.targets.sipeed_tang_primer_20k --build --dock lite` to build a litex soc and after running it the memory init failed.

the following logs where printed:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec  4 2023 21:53:08
 BIOS CRC passed (464b4d02)

 LiteX git sha1: a4ead5ca

--=============== SoC ==================--
CPU:            VexRiscv @ 48MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          128.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:       128.0MiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000| delays: -
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b02 delays: -
  m1, b00: |00000000| delays: -
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b02 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

I also saw that somebody had the same issue in #1750 and it sounded like it was fixed? it still doesn't work for me though.

My board uses the following ram chip:
[image]

trabucayre, 2023-12-05:
Hi,
Could you check **litex-boards** and **litedram** repo version. I see **litex** repo is up-to-date.
Just checked with the same board/dock/DDR3 and memtest is ok.

Glowman554, 2023-12-05:
My litex-boards version is: ba86121 (latest master)
My litedram version is: e835544 (latest master)

trabucayre, 2023-12-05:
What is your gowin toolchain version?
It's weird I have used exactly same repositories version (and double checked with a `pip3 install --user -e .` for all of them).
Could you also check if python uses this repositories (ie doesn't refer to another location/version) ?

Glowman554, 2023-12-05:
Im using Gowin V1.9.8.11 Education not sure how to check what python is using

trabucayre, 2023-12-05:
I use V1.9.9Beta-4 Education (not sure to have tested with 1.9.8).
To know repo location (it's for Linux and don't know how to do the same thing for windows): `pip3 show repoName` where **repoName** is *litex*, *litex-boards*, *litedram*.

Glowman554, 2023-12-05:
```
janick@DESKTOP-QL9SL4A:~$ pip3 show litex
Name: litex
Version: 2023.8
Summary: Python SoC/Core builder for building FPGA based systems.
Home-page: http://enjoy-digital.fr
Author: Florent Kermarrec
Author-email: florent@enjoy-digital.fr
License: BSD
Location: /home/janick/litex
Requires: migen, packaging, pyserial, requests
Required-by: litedram, liteeth, liteiclink, litepcie, litex-boards
janick@DESKTOP-QL9SL4A:~$ pip3 show litex-boards
Name: litex-boards
Version: 2023.8
Summary: LiteX supported boards
Home-page: http://enjoy-digital.fr
Author: Florent Kermarrec
Author-email: florent@enjoy-digital.fr
License: BSD
Location: /home/janick/litex-boards
Requires: litex
Required-by:
janick@DESKTOP-QL9SL4A:~$ pip3 show litedram
Name: litedram
Version: 2023.8
Summary: Small footprint and configurable DRAM core
Home-page: http://enjoy-digital.fr
Author: Florent Kermarrec
Author-email: florent@enjoy-digital.fr
License: BSD
Location: /home/janick/litedram
Requires: litex, pyyaml
Required-by:
janick@DESKTOP-QL9SL4A:~$
```

I will see if the new gowin version makes a difference

Glowman554, 2023-12-05:
Still the same issue with  V1.9.9Beta-4 Education

trabucayre, 2023-12-05:
Ok: I assume LiteX related repos are downloaded in your home directory so everything seems fine here.
Again it's weird...
Could you try with this file?
[sipeed_tang_primer_20k.zip](https://github.com/enjoy-digital/litex/files/13555954/sipeed_tang_primer_20k.zip)

Glowman554, 2023-12-05:
The repos are downloaded in my home directory.
Still the same memtest error with the file you provided though

trabucayre, 2023-12-05:
I must admit I have no idea here: this file work with my board...

Glowman554, 2023-12-05:
Did you check if it uses the same ram chip? In the other issue it sounded like there are differences between the boards that use different ram chips

trabucayre, 2023-12-05:
Yes it's exactly the same RAM chip. It's also the same dock.
My board is 1.5 years old, maybe something as changed.
Have you tried with sipeed example? It's to check if everything is working on your board.

Glowman554, 2023-12-05:
Yep I tried that example and it works

trabucayre, 2023-12-05:
Maybe it's required to ask @sipeed for potential changes in PCB between mine and your board.

Glowman554, 2023-12-06:
Anything new?

trabucayre, 2023-12-06:
Maybe you can ask @sipeed but sending a direct message using twitter.

Glowman554, 2023-12-11:
Anything new?

trabucayre, 2023-12-12:
Integrating the gowin's blob is not straighforward.
I think improving existing support is a better way (and may be used with open source toolchain).
For your specific case it's hard to fix an issue when it's not possible to reproduce locally. Maybe trying to change delay is the first step.

Glowman554, 2023-12-12:
Where would I change them?

Glowman554, 2024-02-15:
I just tested with the newest litex version and it still does not work with my board

enjoy-digital, 2026-10-21:
Closing since there was no further feedback after the update. Feel free to reopen if issues remain with current LiteX/LiteDRAM/litex-boards.

closed 2026-10-21 by enjoy-digital