enjoy-digital/litex #1854: Upgrade of the CV32E40P to the openhwgroup repo ?

pcotret, 2023-12-19:
The CV32E40P is based on the Antmicro fork which is more than 3 years old (https://github.com/litex-hub/pythondata-cpu-cv32e40p/blob/master/pythondata_cpu_cv32e40p/__init__.py).

Would it be interesting to upgrade it with the main openhwgroup implementation ? (https://github.com/openhwgroup/cv32e40p)

I may be afraid that, in 3 years, several features have appeared.
As the interface is mostly similar to the OpenHW CV32E41P, it shouldn't take too long. I just want to check that there is no residual bugs when using the existing CV32E41P :wink:

enjoy-digital, 2024-01-08:
Hi @pcotret,

this would indeed be interesting to use upstream version. Before doing the change, we would need to test current wrapper with upstream CV32E40P and do eventual updates/fixes. If/when working, we could then update the pythondata-cpu-cv32e40p to use this. Do you have the bandwidth/time to do this test? If not, I could still try to do it.

pcotret, 2024-02-07:
Well, as I've worked on the 41P, this should not be an issue. I'll try it this weekend.

nuntipat, 2024-05-13:
Hi @enjoy-digital and @pcotret, I've updated the core to use the upstream version in pull request #1951. Could you please help check it out?

pcotret, 2024-05-13:
Hi @nuntipat, happy to find something who had time to work on it finally!

nuntipat, 2024-05-13:
Hi @pcotret, Do you recall why the value 0x7FFF0880 was used in crt0.S of the CV32E41P and the CV32E40P cores? I think the correct value should be 0xFFFF0880, but I'd like to check with you first.

https://github.com/enjoy-digital/litex/blob/2613ae606a60d250d42089623556a116a33ad0e3/litex/soc/cores/cpu/cv32e41p/crt0.S#L118-L122

pcotret, 2024-05-13:
I had a look back in [my notes](http://pcotret.gitlab.io/blog/processor_in_litex/#interrupt-controller) and the [MIE register documentation](https://docs.openhwgroup.org/projects/cv32e41p-user-manual/control_status_registers.html#machine-interrupt-enable-register-mie).

`0x7FFF0880` is written in the MIE register.

[image]

# In the old Antmicro fork
https://github.com/antmicro/cv32e40p/blob/master/rtl/riscv_cs_registers.sv#L619
```verilog
mie_n.irq_fast     = csr_wdata_int[CSR_MFIX_BIT_HIGH:CSR_MFIX_BIT_LOW];
``` 
https://github.com/antmicro/cv32e40p/blob/master/rtl/include/riscv_defines.sv#L275
```verilog
parameter int unsigned CSR_MFIX_BIT_HIGH = 30;
```

# In the OpenHW repo
https://github.com/openhwgroup/cv32e40p/blob/master/example_tb/core/cv32e40p_tb_subsystem.sv#L63
```verilog
logic [15:0] irq_fast;
```
https://github.com/openhwgroup/cv32e40p/blob/master/example_tb/core/cv32e40p_tb_subsystem.sv#L109
```verilog
.irq_i    ({irq_fast, 4'b0, irq_external, 3'b0, irq_timer, 3'b0, irq_software, 3'b0}),
```

# Summary
- Yes, you're right. When I pushed the 41P, I kept the software stack of the Antmicro 40P where we had only 15 fast interrupts.
- However, the `MSIE` bit (Machine Software Interrupt Enable) is still disabled. Not sure if it's normal for this one.

nuntipat, 2024-05-13:
Thank you for the detailed clarification. I will update my pull request accordingly.

pcotret, 2024-05-23:
Hi @nuntipat @enjoy-digital,
as far as I understand, support for the OpenHwGroup 40P has been done.

@nuntipat also pushed a modification for the URL in https://github.com/litex-hub/pythondata-auto/commit/a9ff1bd3ef9fc21ec7661a36b24a963533394197

Is there another PR to do to update https://github.com/litex-hub/pythondata-cpu-cv32e40p ?

nuntipat, 2024-05-24:
Hi @pcotret, I believe we just need to wait for the litex-hub/pythondata-auto repo's CI to run, which will automatically update the litex-hub/pythondata-cpu-cv32e40p repo.

@enjoy-digital Could you provide an update on the frequency of CI runs? I believe it is currently disabled due to inactivity.

nuntipat, 2024-06-17:
Hi @enjoy-digital, any update on the CI? I check the [litex-hub/pythondata-cpu-cv32e40p](https://github.com/litex-hub/pythondata-cpu-cv32e40p) repo and the last update was still 2 years ago.