enjoy-digital/litex #1857: Failed to modify “boot.c” and the riscv64-unknown-elf/bin/ld report an error “undefined reference to `__heap_start'”

May15356, 2023-12-22:
I tried to modify litex/litex/soc/software/bios/boot.c where I simply insert
```
 int *merged = malloc(sizeof(int));
```
Then the riscv64-unknown-elf/bin/ld report an error “undefined reference to `__heap_start'”
```
/usr/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: ../libc/libc.a(libc_picolib_picosbrk.c.o): in function `sbrk':
/home/may/linux-on-litex-vexriscv/build/sim/software/libc/../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/picolib/picosbrk.c:46: undefined reference to `__heap_start'
/usr/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/may/linux-on-litex-vexriscv/build/sim/software/libc/../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/picolib/picosbrk.c:46: undefined reference to `__heap_start'
/usr/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/may/linux-on-litex-vexriscv/build/sim/software/libc/../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/picolib/picosbrk.c:49: undefined reference to `__heap_end'
/usr/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/may/linux-on-litex-vexriscv/build/sim/software/libc/../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/picolib/picosbrk.c:49: undefined reference to `__heap_end'
/usr/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: ../libc/libc.a(libc_picolib_picosbrk.c.o):/home/may/linux-on-litex-vexriscv/build/sim/software/libc/../../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/picolib/picosbrk.c:41: undefined reference to `__heap_start'
collect2: error: ld returned 1 exit status
```
I'll be very grateful if someone could help!

Mark1626, 2023-12-29:
malloc needs `__heap_start` and `__heap_end` to be defined. This issue https://github.com/enjoy-digital/litex/issues/1466#issuecomment-1285666926 has an example on how to define the heap.

enjoy-digital, 2024-01-12:
Thanks @Mark1626 for the help/answer! We can probably close.

closed 2024-01-12 by enjoy-digital

May15356, 2024-02-21:
Thanks! I have successfully modified!

enjoy-digital, 2024-02-21:
@May15356: Great, thanks for the feedback.